OpenKinect/libfreenect2 #230: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC

NihalRaj, 2015-05-18:
/usr/bin/ld: /home/anubhav/Downloads/espresso-5.1.2/GPU/..//BLAS/blas.a(dgemm.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/home/anubhav/Downloads/espresso-5.1.2/GPU/..//BLAS/blas.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[3]: *** [shared] Error 1
make[3]: Leaving directory `/home/anubhav/Downloads/espresso-5.1.2/phiGEMM/src'
make[2]: *** [phigemm] Error 2
make[2]: Leaving directory `/home/anubhav/Downloads/espresso-5.1.2/phiGEMM'
make[1]: *** [libphiGEMM] Error 2
make[1]: Leaving directory `/home/anubhav/Downloads/espresso-5.1.2/GPU/install'
make: *** [libphiGEMM] Error 2

I am getting this error when i run makefile to compile.
Code used:-
make -f Makefile.gpu pw-gpu 

Please suggest solution. I am using Ubuntu 14.04.

floe, 2015-05-18:
This sounds quite similar to #36 - can you check if maybe your BLAS installation is damaged, or the link to libBLAS.so is somehow missing?

NihalRaj, 2015-05-19:
Thanks,
But problem is not solved yet. I re-install BLAS. Again getting same error.

floe, 2015-05-19:
After having a closer look, are you sure you are actually working on/with libfreenect2? You seem to be working on a different project entirely.

closed 2015-05-19 by floe