OpenKinect/libfreenect2 #232: Ubuntu14.04 : no device connected or failure opening the default one

hiwasawa, 2015-05-18:
hiwasawa commented 4 minutes ago
Hi Now I trying to run Protonect the following environment.

-----My Development Environment----------------

Kinect V2
Ubuntu14.04
USB2.0

$ uname -rv
3.16.0-37-generic #51~14.04.1-Ubuntu SMP Wed May 6 15:23:14 UTC 2015

$ lspci |grep USB
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)

$ glxinfo | grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVS 4200M/PCIe/SSE2
OpenGL core profile version string: 4.4.0 NVIDIA 331.113
OpenGL core profile shading language version string: 4.40 NVIDIA via Cg compiler
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.4.0 NVIDIA 331.113
OpenGL shading language version string: 4.40 NVIDIA via Cg compiler
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 NVIDIA 331.113 331.113
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.00
OpenGL ES profile extensions:

So, I know that microsoft say kinect v2 needs USB3.0.
but The libfreenect2 don't use KinectSDK,
and libusb is a wide use driver.
So, I think libfreenect2 can get some datas from Kinect v2 with usb2.0.

-----Execution result protonect----------------
$ sudo ./bin/Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @1:8 with serial 509608742542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -6: LIBUSB_ERROR_BUSY
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @1:8!

no device connected or failure opening the default one!

How to get some datas from kinect V2.
and if I must use USB3.0.
please tell me why I can't use USB2.0.

xlz, 2015-05-18:
Kinect v2 requires a minimum USB bandwidth of 100MB/s.

hiwasawa, 2015-05-18:
thanks xlz.
ok, so, on my laptop, I can′t  use kinect v2?

Can I use usb3.0 converter ?
http://ca.startech.com/m/HDD/Adapters/USB-3-to-SATA-or-IDE-Hard-Drive-Adapter-Converter~USB3SSATAIDE

xlz, 2015-05-18:
It would not necessarily work. The known working USB 3 controllers are not many. Converters usually add in more complexity.

hiwasawa, 2015-05-18:
Ohh,what would be a good way to do it?

hiwasawa, 2015-05-18:
There are 3 usb ports and a SATA port on my laptop.

xlz, 2015-05-18:
Use a machine with USB 3 ports.

closed 2015-05-20 by floe