OpenKinect/libfreenect2 #233: Preparing for a 0.1 release?

floe, 2015-05-19:
libfreenect2 has been brewing for a while now, and I think we might consider a 0.1 release sometime soon. I'd like to use this issue to collect/discuss all the other issues/PRs that still needs sorting out, please comment below. /cc @christiankerl @xlz @larshg @hovren @gaborpapp @goldhoorn 

Issues/PRs to be addressed before 0.1 release, in no particular order:
- [ ] #223 registration-related stuff
- [ ] #68 FindLibrary files
- [ ] #221 improved stream parsers
- [ ] #151 getting rid of OpenCV

hovren, 2015-05-19:
I have mostly been interested in getting the timestamp data in PR #221 merged, so I am happy as soon as that happens. I have not had the time to look at any of the other PRs or issues in a while, unfortunately. 

I think having a 0.1 release out the door would be really great! For example, I already have prototype code to have KinectFusion in PCL running with libfreenect2, but haven't had time to get it in working order yet. Having an actual libfreenect2 release would probably increase the chances of that code being merged there as well.

goldhoorn, 2015-05-19:
For me i still waiting for the merge. I would really like to integrate this library into http://www.rock-robotics.org/ but for this i need a version that compiles without doing strange 'magic' in the cmake system. The pure library, this is #68 for which is the version i currently using on my test-setup which is not published

gaborpapp, 2015-05-19:
Handling the race condition in #212 solved some crash issues for me at exit. Would be nice to be included.

floe, 2015-05-19:
Since one of the big features of libfreenect2 is the ability to use multiple cameras on one machine, #82 would probably also be nice to include.

xlz, 2015-05-19:
Release needs good stability and documentation:

There are a bunch of crashes on Mac OS. Someone needs to properly test it or diagnose it.

We also need to improve the documentation:
* Incorporate platform specific build instructions and workarounds in documentation: #36 Ubuntu libturbojpeg.so workaround, #109 OpenCL setup, #165 Mac OS how to download cl.hpp, #188 lack of build instructions on Windows, #193 usb permission with udev rules, #194 windows build instructions (previous attempt at windows build documentation #74 #208).
* Add a hardware support list: #146 ASMedia 1142, #216 ASMedia 1042
* Installation linking issue #101 

@goldhoorn #68 mixes too many things on Windows. It is probably a good idea to merge your pkg-config patch first for Linux.

goldhoorn, 2015-05-20:
> @goldhoorn #68 mixes too many things on Windows. It is probably a good idea to merge your pkg-config patch first for Linux.

i still don't get why #68 should be a mix, it is only a cleanup/extension how it is done currently.
As the patch was created @floe and me made sure that the 'old way' of handling librarys is not changed. The new part only uses pure cmake/pkg-config to find librarys. So it is not a mix between things

wiedemeyer, 2015-05-20:
I would love to see the issue #101 to be resolved before the release, so that the library can be compiled and used without modifying the CMakeLists.txt manually.

floe, 2015-11-06:
I guess we're nearly there... I'll now tag the current state as 0.1-rc1. Let's see how this goes.

floe, 2015-11-06:
Here we are: https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1-rc1

steppobeck, 2015-11-06:
Congratulations :-)!!!

--

Stephan Beck
Virtual Reality Systems Group
Faculty of Media
Bauhaus-Universitaet Weimar
Bauhausstra�e 11
99423 Weimar
Germany
phone: +49 3643 - 58 3734
fax  : +49 3643 - 58 3709
email: stephan.beck@uni-weimar.de
email: stephan@pixelstars.de
web  : http://www.uni-weimar.de/de/medien/professuren/vr/

RyanGordon, 2015-11-09:
Awesome!

floe, 2015-12-03:
Once #486 has been resolved/merged, I'd say we're ready for 0.1-final. Please test extensively in the meantime :-)

xlz, 2015-12-03:
And #462.

floe, 2015-12-17:
A couple of other things have crept in, but I think we're now ready for 0.1-rc2, which will then become 0.1-final. I'll create a tag.

xlz, 2015-12-17:
One last thing I can think of other than fixing new bugs is to check memory leaks.

floe, 2015-12-17:
https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1-rc2

Only issue left in 0.1 is #109, which can't be fixed in the repository. I'll look into beignet packaging next.

xlz, 2015-12-24:
### Test plan

#### Build tests

* Full build. `cmake`.
* Optional build. `cmake -DENABLE_OPENGL=OFF`, `cmake -DENABLE_OPENCL=OFF`, `cmake -DENABLE_CXX11=ON`, `cmake -DBUILD_SHARED_LIBS=OFF`.
* Third-party application build. `cmake -DCMAKE_INSTALL_PREFIX=/tmp/freenect2 -DENABLE_EXAMPLES=OFF; make; make install; cd ../examples/build; cmake -Dfreenect2_DIR=/tmp/freenect2/lib/cmake/freenect2`

Expected: `make; ./bin/Protonect` works correctly.

#### Functionality tests

Test under full build.

* Smoke test. `./Protonect`.
* Selecting devices with serial number. `./Protonect 1234567890`.
* Selecting processors. `./Protonect gl`, `./Protonect cl`, `./Protonect cpu`.
* Disabling the viewer. `./Protonect gl -noviewer`, `./Protonect cl -noviewer`.
* Setting log destination. `LOGFILE=test.log ./Protonect`.
* Pausing. `pkill -USR1 Protonect` a few times.
* Restarting the device. Exit and rerun Protonect immediately a few times. Expected: The first few runs report no device found. The last run reports 0x09000 status, waits a few seconds and works correctly.

#### Platform tests

Report details about USB 3 host controller to collect a list of working hardware.

Test combinations of the following:

* Windows 7
* Windows 8
* Debian testing
* Ubuntu 14.04
* Mac OS X

And GPUs

* Intel
* Intel/Nvidia dual
* AMD
* Tegra K1
* Mali (optional)
* Tegra X1 (optional)

Report the hardware combinations.

ahundt, 2015-12-27:
You may want to also consider a "burn in" test, where you just run the driver for a "long time" (perhaps a few days to a week?) to ensure it doesn't fail.

xlz, 2016-01-04:
Making a binary release, especially for Windows is a release target too. More in #244.

xlz, 2016-01-04:
Copyright audit. A list of foreign files and their copyrights:
* `include/internal/CL/cl.hpp`, Copyright (c) 2008-2015 The Khronos Group Inc. (~MIT)
* `src/tinythread/*`, Copyright (c) 2010-2012 Marcus Geelnard. (BSD?)
* `src/flextGL.*`, Copyright (C) 2011 by Thomas Weber. (MIT)
* `Doxyfile.in`, generated from `doxygen -g`, [GPL](https://bugzilla.gnome.org/show_bug.cgi?id=755135)?! Probably need to strip the comments there.
* `src/openni/*`, Copyright (c) Benn Snyder. (Apache/GPL2)

RyanGordon, 2016-01-18:
Should we go ahead and do a 0.1-RC3 or a 0.1 final release? Seems like the only thing left is getting a binary release completed

xlz, 2016-01-18:
Depending on how much testing is still wanted. It doesn't seem to be lots of people doing testing right now. We can release a 0.1.0 quickly, and release 0.1.z as bug fix releases, while adding new features in 0.2.0. Frequent releases seem a good thing for an open source project.

Binary release for windows is done. Linux or MacOSX's homebrew needs outside packagers to do something.

RyanGordon, 2016-01-18:
Agreed, I think getting 0.1.0 out there would be great so we can work on 0.2.0 sooner - I don't think the community expectations are as high for a 0.x than a 1.x

I will look into making a brew PR for libfreenect2 and submitting it

xlz, 2016-01-19:
Also, don't forget to update the docs. There are some changes since the last generated one.

floe, 2016-01-19:
Docs have been updated.

I've been reviewing the diff to v0.1-rc2 and everything looks good, I came across one change I'd like to clarify:

```diff
diff --git a/freenect2.pc.in b/freenect2.pc.in
index 50e5570..0e72ecd 100644
--- a/freenect2.pc.in
+++ b/freenect2.pc.in
@@ -5,8 +5,7 @@ includedir=${prefix}/include
 
 Name: freenect2 
 Description: Driver for the Kinect v2 (aka K4W) 
-Version: 
-Requires: @DEPS_PKGCONFIG@
+Version: @PROJECT_VER@
 Libs: -L${libdir} -lfreenect2
 Cflags: -I${includedir}
 ```

Was it intentional to remove the dependency on pkg-config?

I'll do a final live test tomorrow and then release 0.1.0. I'll also start a branch `v0.1` - in fact, I'd like to try and move back a little as a maintainer and just stay responsible for the stable branch. @xlz, would you be willing to become maintainer for the development/master branch?

xlz, 2016-01-19:
It is. `DEPS_PKGCONFIG` is not actually populated by CMake and the way to populate it is not trivial to get right.

I'm happy to take the responsibility of maintenance of the development branch.

floe, 2016-01-20:
Great, glad to hear that. @JoshBlake could you add @xlz as collaborator?

JoshBlake, 2016-01-21:
Invited @xlz as collaborator.

xlz, 2016-01-21:
@JoshBlake Thank you for the invitation.

floe, 2016-01-21:
@xlz release is ready, can I attach your binaries from https://github.com/xlz/libfreenect2/releases/tag/0.1.0-testing ?

floe, 2016-01-21:
BTW, here you go: https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1.0 (binaries to be added ASAP).

closed 2016-01-21 by floe

xlz, 2016-01-21:
@floe Yes go ahead.

ahundt, 2016-01-22:
Congrats, great work guys!

floe, 2016-01-22:
I thought it would be a good idea to create a DOI for the release and register it at https://zenodo.org/ (see also https://guides.github.com/activities/citable-code/). However, I'm not seeing the OpenKinect/libfreenect2 repository in Zenodo. @JoshBlake can you maybe check if there are any organization settings that could have influence on that?

JoshBlake, 2016-01-22:
@floe I created a Zenodo account and linked to GitHub. Unfortunately, it will only automatically create things for new releases (not existing), per https://twitter.com/zenodo_org/status/467385924532576256. I started an update/DOI entry for the 0.1.0 release. (That's when I realized #537 was an issue.) 

Just now re-reading I realized you also registered. I thought you were asking me to. I can continue the process, just have one question. Or you can continue and manage it through your account. I have no preference. Since I linked after, we might check if the Zenodo webhook entry on Github needs to be reset to yours. Maybe tweet Zenodo.

Anyway, for Authors, I'm just using the maintainers, since there seem to be many contributors. What organization affiliation do @floe @christiankerl and @xlz want to list?

floe, 2016-01-22:
I'd like to give it a try, as I might be using Zenodo more in the future. I think the issue here is that I can't access the webhooks (or any settings at all, for that matter) for the libfreenect2 repo, you would probably have to enable that somewhere in the OpenKinect org settings. Once that works, I'll just fix #537 and do a 0.1.1 release in a few days so Zenodo can then pick that up.

JoshBlake, 2016-01-22:
@floe Granted you more access. Can you see https://github.com/OpenKinect/libfreenect2/settings/hooks or https://github.com/OpenKinect/libfreenect2/settings/collaboration?

floe, 2016-01-22:
Yes, thanks. I'll try to re-connect with Zenodo now.

xlz, 2016-01-22:
@JoshBlake My affiliation is on my github profile.

floe, 2016-01-25:
@xlz was there some minor documentation update left for v0.1.1 regarding wait after Freenect2Device::close()?

xlz, 2016-01-25:
@floe No, it's still not clear what the resolution of #539 is. Fixing by code should be better.

floe, 2016-01-30:
OK, v0.1.1 is out and it does now have a DOI, so it can be properly cited: https://doi.org/10.5281/zenodo.45314 If you're using libfreenect2 for your academic research, please consider doing so.