OpenKinect/libfreenect2 #234: how to get Depth Camera's Intrinsics parameters?

wangsen1312, 2015-05-19:
How can I use libfreenect2 to get Depth Camera's Intrinsics parameters?
Can anyone give me some suggestion? Thank you very much!

JuantAldea, 2015-05-19:
Check file libfreenect2.cpp, Freenect2Device::IrCameraParams getIrCameraParams();

bellekci, 2015-05-19:
Are the parameters returned by getIrCameraParams retrieved from the device?
Weren't parameters hardcoded in libfreenect2 which has also caused the offset problem as mentioned #144.

I just went through the master and it seems to me that they are read from the device. Just wanted to confirm to be sure.

floe, 2015-05-20:
The intrinsics are read directly from the device; they only relate to the geometric camera/lens parameters (see also registration.cpp). The stuff discussed in #144 relates to the depth calculation/deconvolution and is indeed still hardcoded.

zardchim, 2015-05-24:
@floe  is there any comment on the accuracy of the camera parameter? I saw there is calibration tool in the ubuntu ros package, do we have similar feature in Windows? or could I do the calibration on my own and use my own parameter instead @@?

floe, 2015-05-24:
A well-done chessboard calibration with an external calibration tool will very likely be of better quality than the factory-programmed calibration. However, it's difficult to directly compare them, as the factory calibration uses a different method for calculating, e.g. distortion, than common packages like OpenCV.

zardchim, 2015-05-26:
In libfreenect2.cpp

zardchim, 2015-05-26:
In libfreenect2.cpp there are functions to get rgb_camera_params_ & ir_camera_params_ in order to do registration, is it possible for us to replace these parameters to those generated from opencv (e.g. from iai_kinect calibration result) ?

floe, 2015-05-26:
It might work for IrCameraParams; the format/algorithm for ColorCameraParams is different from the one OpenCV uses (IrCameraParams is just intrinsic & distortion parameters, ColorCameraParams is a mix of intrinsic & extrinsic parameters).

lakehanne, 2015-07-13:
Oh I see. And this explains #144.

closed 2016-01-27 by xlz