OpenKinect/libfreenect2 #237: Getting all 0's for depth

anshulvj, 2015-05-20:
I'm running the protonect example on Ubuntu 15.04 with 3.19.0-15-generic kernel. lsusb output is as follows:

         Bus 004 Device 002: ID 8087:8000 Intel Corp. 
         Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
         Bus 003 Device 002: ID 8087:8008 Intel Corp. 
         Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
         Bus 002 Device 007: ID 045e:02c4 Microsoft Corp. 
         Bus 002 Device 006: ID 045e:02d9 Microsoft Corp. 
         Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
         Bus 001 Device 005: ID 413c:2111 Dell Computer Corp. 
         Bus 001 Device 004: ID 046d:c077 Logitech, Inc. 
         Bus 001 Device 007: ID 0cf3:0036 Atheros Communications, Inc. 
         Bus 001 Device 009: ID 045e:02d9 Microsoft Corp. 
         Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lshw output for usb is as follows:

       *-usb:0
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB xHCI
             vendor: Intel Corporation
             physical id: 14
             bus info: pci@0000:00:14.0
             version: 05
             width: 64 bits
             clock: 33MHz
             capabilities: xhci bus_master cap_list
             configuration: driver=xhci_hcd latency=0
             resources: irq:27 memory:f7e00000-f7e0ffff
      *-usb:1
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB EHCI #2
             vendor: Intel Corporation
             physical id: 1a
             bus info: pci@0000:00:1a.0
             version: 05
             width: 32 bits
             clock: 33MHz
             capabilities: ehci bus_master cap_list
             configuration: driver=ehci-pci latency=0
             resources: irq:16 memory:f7e1c000-f7e1c3ff
      *-usb:2
             description: USB controller
             product: 8 Series/C220 Series Chipset Family USB EHCI #1
             vendor: Intel Corporation
             physical id: 1d
             bus info: pci@0000:00:1d.0
             version: 05
             width: 32 bits
             clock: 33MHz
             capabilities: ehci bus_master cap_list
             configuration: driver=ehci-pci latency=0
             resources: irq:23 memory:f7e1b000-f7e1b3ff

I'm not sure what's going on. I've definitely connected the Kinect into a usb 3.0 port and compilation was no issue at all. I'm getting the rgb image all right, but nothing on the ir, depth and registration (all black). Any help/pointers will be appreciated.

wiedemeyer, 2015-05-20:
What is the output of `Protonect`, does it say something like `failed to submit transfer`? Is the Kinect listed as SuperSpeed in `dmesg`?

anshulvj, 2015-05-20:
No I don't get any failure messages when I run Protonect. The output for Protonect is as follows:

    anshulvj@anshulspc:~/KinectV2Software/libfreenect2/examples/protonect$ sudo ./bin/Protonect
    [Freenect2Impl] enumerating devices...
    [Freenect2Impl] 12 usb devices connected
    [Freenect2Impl] found valid Kinect v2 @2:9 with serial 031084540347
    [Freenect2Impl] found 1 devices
    [Freenect2DeviceImpl] opening...
    [Freenect2DeviceImpl] opened
    [Freenect2DeviceImpl] starting...
    [Freenect2DeviceImpl] ReadData0x14 response
    92 bytes of raw data
    0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41        2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
    0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00       2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
    0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
    0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
    0x0040:  31 33 00 00 00 02 0e 04 47 4b 50 32 31 30 2e 31    31332e2e2e2e2e2e474b503231302e31
    0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

    [Freenect2DeviceImpl] ReadStatus0x090000 response
    4 bytes of raw data
    0x0000:  01 26 00 00                                        2e262e2e

    [Freenect2DeviceImpl] ReadStatus0x090000 response
    4 bytes of raw data
    0x0000:  03 26 00 00                                        2e262e2e

    [Freenect2DeviceImpl] enabling usb transfer submission...
    [Freenect2DeviceImpl] submitting usb transfers...
    [Freenect2DeviceImpl] started
    device serial: 031084540347
    device firmware: 4.1.3911.0.7
    [DepthPacketStreamParser::onDataReceived] not all subsequences received 0

Regarding dmesg, I'm not sure how to read the output, since I've never used it before. But the Kinect is listed in a lot of places, both as high-speed and SuperSpeed. I'm pasting all the output from the long output message, that I think is relevant to the Kinect:
......
[   11.524566] usb 1-1: new high-speed USB device number 3 using xhci_hcd
[   11.655053] usb 1-1: New USB device found, idVendor=045e, idProduct=02d9
[   11.655066] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   11.655068] usb 1-1: Product: NuiSensor Adaptor      
[   11.655069] usb 1-1: Manufacturer: Microsoft Corporation  
[   11.655684] hub 1-1:1.0: USB hub found
[   11.655920] hub 1-1:1.0: 1 port detected
.......
[55011.081110] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[55011.098789] usb 2-1: New USB device found, idVendor=045e, idProduct=02d9
[55011.098792] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[55011.098794] usb 2-1: Product: NuiSensor Adaptor      
[55011.098795] usb 2-1: Manufacturer: Microsoft Corporation  
[55011.099375] hub 2-1:1.0: USB hub found
[55011.099519] hub 2-1:1.0: 1 port detected
[55011.285121] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[55011.415650] usb 1-1: New USB device found, idVendor=045e, idProduct=02d9
[55011.415655] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[55011.415658] usb 1-1: Product: NuiSensor Adaptor      
[55011.415660] usb 1-1: Manufacturer: Microsoft Corporation  
[55011.416289] hub 1-1:1.0: USB hub found
[55011.416524] hub 1-1:1.0: 1 port detected
[55012.771241] usb 2-1.1: new SuperSpeed USB device number 5 using xhci_hcd
[55012.787200] usb 2-1.1: New USB device found, idVendor=045e, idProduct=02c4
[55012.787203] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[55012.787205] usb 2-1.1: Product: Xbox NUI Sensor
[55012.787206] usb 2-1.1: Manufacturer: Microsoft
[55012.787208] usb 2-1.1: SerialNumber: 031084540347
[56015.999150] usb 2-1.1: reset SuperSpeed USB device number 5 using xhci_hcd
[56016.014599] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ee62ea80
[56016.014603] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ee62eab0
[56016.014604] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ee62eb10
[56016.014605] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ee62eae0
[56016.362861] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 11 ep 2 with no TDs queued?
[... 30 lines cut ...]
[56640.455507] usb 2-2: new SuperSpeed USB device number 8 using xhci_hcd
[56640.640748] usb 1-2: New USB device found, idVendor=045e, idProduct=02d9
[56640.640755] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[56640.640759] usb 1-2: Product: NuiSensor Adaptor      
[56640.640763] usb 1-2: Manufacturer: Microsoft Corporation  
[56640.641577] hub 1-2:1.0: USB hub found
[56640.641787] hub 1-2:1.0: 1 port detected
[56643.436564] usb 2-2: New USB device found, idVendor=045e, idProduct=02d9
[56643.436567] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[56643.436570] usb 2-2: Product: NuiSensor Adaptor      
[56643.436571] usb 2-2: Manufacturer: Microsoft Corporation  
[56643.437205] hub 2-2:1.0: USB hub found
[56643.437376] hub 2-2:1.0: 1 port detected
[56643.905892] usb 2-2.1: new SuperSpeed USB device number 9 using xhci_hcd
[56643.921593] usb 2-2.1: New USB device found, idVendor=045e, idProduct=02c4
[56643.921596] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[56643.921597] usb 2-2.1: Product: Xbox NUI Sensor
[56643.921599] usb 2-2.1: Manufacturer: Microsoft
[56643.921600] usb 2-2.1: SerialNumber: 031084540347
[... cut ...] device number 9 using xhci_hcd
[56670.575177] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep e01e7a80
[56670.575180] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep e01e7ab0
[56670.575181] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep e01e7b10
[56670.575182] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep e01e7ae0
[56670.923896] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 18 ep 2 with no TDs queued?

xlz, 2015-05-20:
Everything looks normal. What is the further output of Protonect? (using OpenGL?) And your GPU?

anshulvj, 2015-05-20:
anshulvj@anshulspc:~/KinectV2Software/libfreenect2/examples/protonect$ sudo ./bin/Protonect 
[sudo] password for anshulvj: 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 031084540347
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 02 0e 04 47 4b 50 32 31 30 2e 31    31332e2e2e2e2e2e474b503231302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 031084540347
device firmware: 4.1.3911.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1020
[DepthPacketStreamParser::onDataReceived] image data too short!
[DepthPacketStreamParser::onDataReceived] image data too short!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 767
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
...
(after I hit escale)
...
[Freenect2DeviceImpl] stopping...  
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
.
.
.
.
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] already closed, doing nothing

I checked the code but I wasn't able to figure out whether it's using the GPU or the CPU (Is there any knob in the example to change this?). Also, I'm not sure about OpenGL but I do have the version OpenGL version string: 3.0 Mesa 10.5.2 on my machine. I'm presuming it wouldn't have run/installed if there was a dependency(?). Pardon my less knowledge, it's the first time I'm using Kinect on linux.

xlz, 2015-05-20:
```
...
(after I hit escale)
```
Crucial information is in here. You need to find out which of OpenGL and OpenCL depth processor is being used.

OpenGL has bugs that output black. OpenCL will output blank sometime if /sys/module/i915/parameters/enable_cmd_parser=1.

floe, 2015-05-22:
Try `Protonect cl` or `Protonect cpu` with the latest master branch. If one of those works, then it's your OpenGL implementation which is causing problems. Otherwise, it's an USB3 issue.

anshulvj, 2015-05-27:
Hi floe,

I tried running with both ./Protonect cl and ./Protonect cpu but both gave me the same black depth frame. The outputs were as follows:
for ./Protonect cl:

anshulvj@anshulspc:~/KinectV2Software/libfreenect2/examples/protonect/bin$ sudo ./Protonect cl
[sudo] password for anshulvj: 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 031084540347
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 02 0e 04 47 4b 50 32 31 30 2e 31    31332e2e2e2e2e2e474b503231302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 031084540347
device firmware: 4.1.3911.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[OpenGLDepthPacketProcessor] avg. time: 4.14723ms -> ~241.125Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.4655ms -> ~51.3729Hz
[OpenGLDepthPacketProcessor] avg. time: 3.96573ms -> ~252.161Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.8704ms -> ~52.9931Hz
[OpenGLDepthPacketProcessor] avg. time: 4.20689ms -> ~237.705Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.1754ms -> ~52.1501Hz
[OpenGLDepthPacketProcessor] avg. time: 4.01736ms -> ~248.92Hz
^C[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
.
.
.

And for "./Protonect cpu" command:

anshulvj@anshulspc:~/KinectV2Software/libfreenect2/examples/protonect/bin$ sudo ./Protonect cpu
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 031084540347
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 02 0e 04 47 4b 50 32 31 30 2e 31    31332e2e2e2e2e2e474b503231302e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 031084540347
device firmware: 4.1.3911.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 768
[OpenGLDepthPacketProcessor] avg. time: 4.07567ms -> ~245.359Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.8983ms -> ~52.9148Hz
[OpenGLDepthPacketProcessor] avg. time: 3.95266ms -> ~252.994Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.058ms -> ~52.4713Hz
[OpenGLDepthPacketProcessor] avg. time: 3.99991ms -> ~250.006Hz
[TurboJpegRgbPacketProcessor] avg. time: 19.073ms -> ~52.4302Hz
[OpenGLDepthPacketProcessor] avg. time: 3.89306ms -> ~256.867Hz
[TurboJpegRgbPacketProcessor] avg. time: 18.5482ms -> ~53.9136Hz
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
.
.
.

You said "...Otherwise it's an USB3 issue". If this is the case, then what's the solution? 

Thanks

larshg, 2015-05-27:
@anshulvj  did you pull latest master? because it print out from OpenGLdepthprocesser for both runs, hence it haven't run with either the OpenCL or the CPU implementation.

You write you have opengl 3.0 ? I think opengl 3.3 is necessary.

anshulvj, 2015-05-27:
Wow! I pulled the latest master and using it with the cpu argument worked like charm!!!! Thank you all for your responses and helping me get this running.

I'm guessing now that this (previously an) issue was with me using an integrated graphics card rather than a full fledged GPU?

xlz, 2015-05-28:
Looks like the old bug #117.

wiedemeyer, 2015-05-28:
It should also work with integrated GPUs. For OpenGL you need at least version 3.3 as @larshg said. And for OpenCL you need to install the `opencl-headers` and drivers for your GPU, in case of Intel you will have to install beignet, [here are some instructions](https://github.com/code-iai/iai_kinect2#opencl-with-intel-gpu).

horxianfeng, 2015-05-30:
@anshulvj @larshg  i may face the same problem also..can u explain what is the meaning of pull the latest master?

larshg, 2015-05-30:
If some new pull request has mean merged since you last fetched/pulled from the repository, you need to update your local files to get the newest changes - ie. Pull the latest master (branch). Hope it clarifies it :-)

horxianfeng, 2015-05-30:
okie

I am using ubutu 14.04. Ros Indigo.
intel 15. Nvidia 840.

So far, i install libreenect2 succesfully.
However, when i try to run rosrun kinect2_bridge kinect2_bridge, i face the error below( I had tried to update my kernel and etc, but i still cant solve the problem):

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 508442542542
device firmware: 4.3.3912.0.7
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[... 44 lines cut ...]
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.8608093261719, 0, 254.502197265625;
  0, 365.8608093261719, 205.2525939941406;
  0, 0, 1]
distortion coefficients ir:
[0.09640563279390335, -0.2757557928562164, 0, 0, 0.09214892983436584]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052; 0; 0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

/home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge/home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: symbol lookup error: /home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi
: symbol lookup error: /home/xfhor/kinectv2/devel/lib/kinect2_bridge/kinect2_bridge: undefined symbol: _ZN12libfreenect222SyncMultiFrameListener15waitForNewFrameERSt3mapINS_5Frame4TypeEPS2_St4lessIS3_ESaISt4pairIKS3_S4_EEEi

larshg, 2015-05-30:
Hmm, I don't use the bridge nor ros, so you might be better off posting your problem athttps://github.com/code-iai/iai_kinect2 or try run protonect as it's the one I'm familiar with. Maybe @wiedemeyer can help?

closed 2015-06-24 by anshulvj

wenbo001, 2016-10-28:
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:4 with serial 004645263147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [Freenect2DeviceImpl] started
device serial: 004645263147
device firmware: 2.3.3913.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 1022
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 1
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
...
[Info] [DepthPacketStreamParser] 375 packets were lost
Segmentation fault: 11
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]
I could not open it...it quit after the process completed...