OpenKinect/libfreenect2 #242: Cannot find device on Windows 7 x64

zergb, 2015-05-22:
Follow the instruction from https://github.com/OpenKinect/libfreenect2/issues/211, I build and run libfreenect2 on windows 7 x64 and visual studio 2010. The Protonect always output as below. I have used USB 3 port and update the libusbK drivers. The only difference is unchecking OpenCL and OpenGL. Does anyone has idea about this?

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 7 usb devices connected
[Freenect2Impl] found 0 devices
[Freenect2Impl] requested device 0 is not connected!
no device connected or failure opening the default one!

larshg, 2015-05-26:
Is the xbox nui sensor (Composite parent) visible in Device manager? Anyways its not found, so changing the OpenCL/GL/Cpu doesn't really help.

zergb, 2015-05-27:
It's visible and exact the same name. Also I've tried different version of 3.0.6.0 and 3.0.7.0. The application can tell device number change when i plug/unplug it. But cannot init it.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe