OpenKinect/libfreenect2 #243: OpenNI2 driver

hanyazou, 2015-05-24:
I've roughly ported the OpenNI2FreenectDriver for libfreenect2. While this port is little bit messy, it works with PCL's pcl_openni2_viewer and OpenNI2's NiViewer on my Mac OSX.
Please look at https://github.com/hanyazou/libfreenect2/tree/master/OpenNI2-Freenect2Driver.  I'm glad you give me some ideas of how I merge this work with the project.

floe, 2015-05-24:
Thanks for your work! Just for my understanding, does that mean you can now use the NiTE skeleton tracking module with the Kinect v2?

hanyazou, 2015-05-24:
I'd like to try on it next weekend. But the primesense web site was closed and I could not find NiTE middleware. Do you have correct download link?

jas99, 2015-05-24:
try here 
https://github.com/PrimeSense

gaborpapp, 2015-05-25:
openni2: https://github.com/occipital/openni2
nite2 archive: https://bitbucket.org/kaorun55/openni-2.2/src/2f54272802bfd24ca32f03327fbabaf85ac4a5c4/NITE%202.2%20%CE%B1/?at=master

hanyazou, 2015-05-31:
I've tried the sample executable in the NiTE binary, NiTE-MacOSX-x64-2.2.tar.zip. I did some debugging and added hacking code to my driver because the NiTE required some compatibilities with old Kinect drivers to work correctly. The default depth resolution must be 640x480 and device name must be "Kinect" exactly. ("Kinect 2" does not work for the NiTE.)

Now, that looks working.
[image]

HenningJ, 2015-06-02:
Nice work. I modified your code a little to make it compile on Windows 7 with Visual Studio 2008 (ripped out some c++11 stuff and replaced `sys/time.h` with `boost/chrono`), now it works great for me :-)

HenningJ, 2015-06-08:
After working with this some more, I noticed a couple of problems:

1. Big memory leak. It only ran for a minute or so, before it crashed with a bad_allocation exception. I'm using 32bit instead of 64bit, so my memory is full rather quickly, and you might not have noticed that at all.  
I guess you have to delete the frame pointer in the `libfreenect2::FrameListener::onNewFrame()` method. At least that's [what protonect does](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/frame_listener_impl.cpp#L133), once there is a new frame.
2. When closing the OpenNI device, you [invalidate an iterator](https://github.com/hanyazou/libfreenect2/blob/master/OpenNI2-Freenect2Driver/src/DeviceDriver.cpp#L442) before using it in the following line.

HenningJ, 2015-06-08:
Also, this seems to add a lot of overhead to the libfreenect2 processing. With protonect, I get around 30 fps with the OpenGL depth processor, with this I barely get 10 fps. Maybe too much work is being done in the `onNewFrame()` callback? Maybe you can just store the pointer to the new frame and do the actual creation of a new OpenNI frame in a different thread?

hanyazou, 2015-06-08:
@HenningJ I've fixed onNewFrame() in <a href=https://github.com/hanyazou/libfreenect2/commit/4e8eb54894f96c2a026445bb58635a76512abe0c>4e8eb54894</a>. Please look at this and let me know if you still have the problem. I will investigate the closing problem and the overhead next weekend. Thank you so much for your testing and comments.

HenningJ, 2015-06-09:
You're right, no more memory leak. :+1: 

It would also be nice to turn off the registration (for speed), and a way to get the original resolution (HD) RGB image.

skarard, 2015-06-11:
Hello,

I have attempted to follow what you are doing.

OpenNi2 SimpleViewer complained that the depth and color streams were different sizes. So I was able to change the DepthDriver.cpp to resolve that issue.

You mentioned that NiTE2 looks for "Kinect" and a depth resolution of 640x480.

I then change DepthDriver.cpp resolution settings back to 640x480 I received Illegal instruction (core dumped) from NiTE2 UserViewer.

HenningJ, 2015-06-11:
@hanyazou I played around some more and did the following changes:

* Instead of directly processing the frames in the `onNewFrame()` callback, I'm using the `SyncMultiFrameListener` like in protonect
* the OpenNI frames are created in a separate thread
* the `libreenect2::Registration` is applied to the whole image instead of each pixel individually

Before these changes, I had framerates like the following:

```
[OpenCLDepthPacketProcessor] avg. time: 10.8811ms -> ~91.9023Hz
[TurboJpegRgbPacketProcessor] avg. time: 153.137ms -> ~6.53011Hz
 26017347 VERBOSE    [FPS] Depth: 29.36 Color: 7.39                  <- inside OpenNI
```

Now I got this:

```
[OpenCLDepthPacketProcessor] avg. time: 10.7307ms -> ~93.1904Hz
[TurboJpegRgbPacketProcessor] avg. time: 22.0074ms -> ~45.4393Hz
 22014990 VERBOSE    [FPS] Depth: 29.32 Color: 29.57
```

Much nicer. Now OpenNI isn't slowing down libfreenect2 anymore.
I can do a pull request to your repository tomorrow, if you like.

hanyazou, 2015-06-11:
@HenningJ I like! Please send your improvement to me.

I must merge from upstream too. And I have some bugs. It might take a little time to do them.

hanyazou, 2015-06-13:
I've merged HanningJ's improvement and add switch of enable/disable the registration. Now you can retrieve full HD color image from the driver.

@skarard I can run NiTE UserViewer to track my body on my Mac. Could you explain your environment and problems in detail?

MasWag, 2015-06-14:
Excellent, 

I wrote another OpenNI2 driver. I have also tried Nite but it does not work.
In my implementation, recording works. I don't know whether recoding works in this implementation. If not, I'll add some properties. But I have no Kinect v2 now.So I cannot test it well....

hanyazou, 2015-06-14:
@MasWag Could you tell me what case do you mean by 'recording'? I will try.

MasWag, 2015-06-14:
Recording is Recording feature of OpenNI2.

hanyazou, 2015-06-14:
I pressed 's' key while running the NiViwer to save Captured.oni of 500MB in several seconds. How can I use this file?

MasWag, 2015-06-14:
Open oni file like this
https://gist.github.com/MasWag/294382bd48666356368f

hanyazou, 2015-06-14:
Though I'm not sure that I could build your play_depth correctly, it seems to work. It shows recorded depth movie quickly.

MasWag, 2015-06-14:
Hmm it seems timestamp feature is not implemented.In my implementation, timestamp is added like this (https://github.com/MasWag/OpenNI2/blob/libfreenect2/Source/Drivers/Libfreenect2/LF2DepthStream.cpp#L45)

MasWag, 2015-06-14:
I have wrote timestamp and made pull request. (https://github.com/hanyazou/libfreenect2/pull/2) Please check it.

hanyazou, 2015-06-14:
You are right. The play_depth replay depth movie in correct way if the driver use frame->sequence*33369 as time stamp. The libfreenect2 was improved around time stamp in last weeks If I remember correctly. I will check about this issue next week. Thank you.

hanyazou, 2015-06-14:
@MasWag Thank you for your patch. I've fixed timestamp handling in <a href=https://github.com/hanyazou/libfreenect2/commit/059276b7b3a25d091e3f1e1c696ab7f6a153d6a0>059276b7b3</a>.

MasWag, 2015-06-14:
@hanyazou yes.It convert 30Hz to micro seconds. I think it should be 33333 but when I got the value (interval between frames) of 30fps ASUS Xtion, it was 33369 or sometimes 33370. So I used 33369.

HenningJ, 2015-06-15:
> add switch of enable/disable the registration. Now you can retrieve full HD color image from the driver.

Nice. :+1: 

Just one thing: You don't initialize the `Freenect2Driver::Registration::enabled` bool, so sometimes the `Registration` thought it was enabled by default, even though the `DepthStream` made sure the registration is OFF by default. Maybe you can use `setImageRegistrationMode(ONI_IMAGE_REGISTRATION_OFF);` in the [`DepthStream` constructor](https://github.com/hanyazou/libfreenect2/blob/master/drivers/OpenNI2-Freenect2Driver/src/DepthStream.cpp#L25), instead of setting the setting the `image_registration_mode` variable directly. Then the `Registration` and the `DepthStream` would always agree, whether or not the registration is active. 
Or you can just initialize the `enabled` bool to false. But then you'd have to change two places if you ever wanted to change the default behaviour of the registration.

hanyazou, 2015-06-15:
@HenningJ Thank you for your feedback. Fixed in 90d65b2.

hanyazou, 2015-06-15:
Ah, I found that fixing timestamp handling in 059276b7b3 has broken something for PCL viewer. The pcl_openni2_viewer won't show any point cloud because of the change. Just black screen. I will investigate this problem next weekend.

HenningJ, 2015-06-15:
Btw. libfreenect2 master was changed to use BGRX pixels instead of BGR. I changed the method to copy data into the OpenNi color frames accordingly in d942a72a833bf1d0ff34948a2dd5428c40ea8ff7

hanyazou, 2015-06-16:
Merged in 3ba0532

lrvk123, 2015-06-24:
I have issues compiling the pull-request branch of hanyazou/libfreenect:

In file included from /usr/include/c++/4.8/array:35:0,
                 from /home/ubuntu/Build/openni2-freenect2/libfreenect2/drivers/OpenNI2-Freenect2Driver/src/DeviceDriver.cpp:23:
/usr/include/c++/4.8/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^

skarard, 2015-06-24:
In the CMakeLists.txt file there is a place to put flags for gcc add
-std=c++11 as one of the flags.

lrvk123, 2015-06-24:
Thank you, skarard. I found that MAKEFLAGS do not work and deleted my previous comment. Changing CMakeLists.txt flags was successful.

lrvk123, 2015-06-24:
I wonder if someone could help.
1. I compiled OpenNI2
2. I have compiled the libfreenect2 driver and placed it in OpenNI2/Bin/Arm-Release/OpenNI2/Drivers/libFreenect2Driver.so
3. Compiled pcl with OpenNI2 enabled. 

When I run 'pcl_openni2_viewer'
I get:
ubuntu@tegra-ubuntu:~/Build/pcl-trunk/build$ pcl_openni2_viewer 
terminate called after throwing an instance of 'pcl::IOException'
  what():  void pcl::io::OpenNI2Grabber::setupDevice(const string&, const pcl::io::OpenNI2Grabber::Mode&, const pcl::io::OpenNI2Grabber::Mode&) in /home/ubuntu/Build/pcl-trunk/io/src/openni2_grabber.cpp @ 325 : No devices connected.
Aborted

However, the lsusb shows connected Microsoft device.

Did I miss a step and did not place the compiled driver .so file in a correct folder?

HenningJ, 2015-06-25:
@lrvk123: The first step is always: does protonect work and display all images?
If so, does the OpenNI dirver work with the NiViewer?
Maybe you can find out, where the problem is exactly.

lrvk123, 2015-06-25:
@HenningJ : The Protonect binary displays all streams: ir/rgb/depth. OpenNI2/Bin/Arm-Release/org.openni.Samples.SimpleViewer works as well.
I suppose then that the issue lies with pcl_openni2_viewer?

HenningJ, 2015-06-25:
I guess so. 
You could try to turn on the logging of OpenNi to get more information about what happens behind the scenes. Grab the OpenNi.ini from the Config folder, copy it into `OpenNI2/Bin/Arm-Release/` and un-comment the appropriate lines:

```
Verbosity=0
LogToConsole=1
```

That should tell you where the OpenNi drivers are loaded from and maybe why the libfreenect2 driver couldn't be loaded.

@hanyazou got the pcl_viewer to work, so maybe he can help some more.

hanyazou, 2015-06-25:
@lrvk123 I'm not sure because I have no Linux system. Please check which OpenNI2 library is linked with the pcl_openni2_viewer executable. There might be another libOpenNI2.so and OpenNI2/Drivers/ pair. Try remove another driver directories and make symbolics link to your OpenNI2/Bin/Arm-Release/OpenNI2.

And you might need this patch:

diff --git a/visualization/tools/openni2_viewer.cpp b/visualization/tools/openni2_viewer.cpp
index 595e5a9..00b2b86 100644
--- a/visualization/tools/openni2_viewer.cpp
+++ b/visualization/tools/openni2_viewer.cpp
@@ -296,6 +296,7 @@ main (int argc, char** argv)
   pcl::io::OpenNI2Grabber::Mode image_mode = pcl::io::OpenNI2Grabber::OpenNI_Default_Mode;
   bool xyz = false;

+#if 0
   if (argc >= 2)
   {
     device_id = argv[1];
@@ -341,6 +342,7 @@ main (int argc, char** argv)
       cout << "Device ID not set, using default device: " << device->getStringID () << endl;
     }
   }
+#endif // 0

   unsigned mode;
   if (pcl::console::parse (argc, argv, "-depthmode", mode) != -1)

lrvk123, 2015-06-25:
@HenningJ : Thanks, I got "Using ./OpenNI2/Drivers' as driver path and a confirmation that libFreenect2Driver.so is loaded. The dynamically linked library is in the correct place: 
	libOpenNI2.so => /opt/OpenNI-Linux-Arm-2.3/Redist/libOpenNI2.so (0xb531a000)

@hanyazou : I applied your patch, checked the linker paths and recompiled, the pcl_openni2_viewer starts up now but I get just a black frame. 

The Kinect2  works with Libfreenect2 / Protonect and also with OpenNI2/Bin/Arm-Release/NiViewer so I can guess that the issue could be with the dynamically linked libusb? The linked library of pcl_openni2_viewer is:
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb5380000)
while libFreenect2Driver.so is linked against 	
        libusb-1.0.so.0 => /home/ubuntu/Build/openni2-freenect2/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0xb6b7b000)
Does the pcl_openni2_viewer use the libFreenect2Driver.so's libusb or its own?

The errors are below:
-------------------------------------------------------------------------------------------------
[Freenect2Impl] found valid Kinect v2 @2:4 with serial 084657533947
[Freenect2Impl] found 1 devices
OpenNI2-Freenect2Driver: deiveOpen: freenect2://0
OpenNI2-Freenect2Driver: Opening device freenect2://0
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0a 15 47 48 56 32 32 36 2e 31    31332e2e2e2e2e2e4748563232362e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  23 26 00 00                                        23262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
...========== OMITTED lines ===============
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[Freenect2DeviceImpl] started
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] starting...
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
void pcl::io::openni2::OpenNI2Device::setSynchronization(bool) @ /home/ubuntu/Build/pcl-trunk/io/src/openni2/openni2_device.cpp @ 432 : Enabling depth color synchronization failed: 
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
...======== OMITTED lines
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!

hanyazou, 2015-06-25:
The "depth color synchronization failed" is not a fatal error. You might need to update PCL source code. Please see <a href=https://github.com/PointCloudLibrary/pcl/commit/4068b514d19deb390371c33754dc1f693eba4bf0>
4068b51</a> and <a href=bf601f0c8f67770f5c8b803d72bdee5fe4dc0b51>bf601f0</a> . And it might be better to use the same libusb.

rossifier, 2015-07-08:
Hi all,
I am trying to use @MasWag 's OPENNI freenect2 driver. So I run this SimpleViewer program and I got the same error as @skarard:
Error - expect color and depth to be in same resolution: D: 512x424, C: 1920x1080
But protonect example is running normally.
How can I set the depth size properly so that i can use OpenNI with kienct2? 
As far as I remember in protonect the color and depth stream have not the same size either.
Is there any solution to this issue? 
Thank you for your nice job.

MasWag, 2015-07-08:
Hi rossifier,
I know that error.  SimpleViewer expect color and depth to be in same resolution . OpenNI2 does not require it.  You can use SimpleRead to test if depth works well. 
If you delete resolution chcking in SimpleViewer, it might work. I do not test it.

rossifier, 2015-07-08:
I commented the lines you mentionned. It is still not working. The output of SimpleViewer after commenting these lines is this:
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:4 with serial 504730242542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
After initialization:

[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  03 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 0f 10 47 4d 46 36 33 31 2e 31    31332e2e2e2e2e2e474d463633312e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  5d 26 00 00                                        5d262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  43 26 00 00                                        43262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[Freenect2DeviceImpl] starting...
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[DepthPacketStreamParser::onDataReceived] skipping depth packet
Segmentation fault (core dumped)

The output of SimpleRead is the following
./SimpleRead 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:4 with serial 504730242542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  03 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 0f 10 47 4d 46 36 33 31 2e 31    31332e2e2e2e2e2e474d463633312e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  3f 26 00 00                                        3f262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[23158086]      697
[23224824]      694
[23258193]      697
[23291562]      698
[23324931]      695
[23358300]      696
[23391669]      696
[23425038]      697
[23458407]      696
[23491776]      697
[23525145]      696
^C

MasWag, 2015-07-08:
Hi,

It seems difficult to modify SimpleViewer. You should write viewer.

hanyazou, 2015-07-08:
@rossifier could you use NiViewer instead of SimpleViewer? I'm sure that NiViewer running correctly with the driver.

michaelwolfram, 2015-07-09:
Hello!
Could you guys please give me some hints on how to set things up correctly such that I can use OpenNI2 with Kinectv2, e.g. oni file recording.
I tried some things but could not figure out the right way. I have to use the normal OpenNI2 library, right? And then insert the libfreenect2 openni2 driver in the correct folder?
Protonect itself is running normally.

Thank you,
Michael

rossifier, 2015-07-09:
Hi Michael,

You'll need to get @MasWag OpenNI driver for kinect2, compile it and link your programs to the generated libOpenNI2.so.
Hope it works :)

rossifier, 2015-07-09:
Thank you @hanyazou and @MasWag for your comments. For the depth viewer, I implemented an opencv based viewer. But I still have some problems with displaying texture (see image below which is the ouput of NiViewer) 
[image]
Any suggestions?

rossifier, 2015-07-09:
Actually, I tried to make an opencv based color viewer, so I did the following:

       openni::RGB888Pixel* pColor = (openni::RGB888Pixel*)frame.getData();
        cv::Mat cv_frame;
        cv_frame.create(frame.getHeight(), frame.getWidth(), CV_8UC3);
        memcpy( cv_frame.data, pColor, 3*frame.getHeight()*frame.getWidth()*sizeof(uint8_t) );
        cv::imshow("color", cv_frame);
        cv::waitKey(1);
And I got the same image as displayed by NiViewer:
[image]

michaelwolfram, 2015-07-09:
Ah OK. And my existing OpenNI2 programs will automatically work on the Kinect v2, right?! I cannot use the original libfreenect2 master but have to use MasWag's sequence branch, correct?
I was wondering what the difference between @MasWag and @hanyazou version is. For me the most imporant feature is recording and playback of .oni files since I cannot think of a neater way.

---

Well, I have a similar OpenCV viewer with my ASUS Xtion and I am converting the unsigned char* (=frame.getData()) like this and it does work:
	const uint8_t* colorImageBuffer = (const uint8_t*) _uchColorImage;
	rgbImage.create(cameraInfo.color.height, cameraInfo.color.width, CV_8UC3);
	memcpy(rgbImage.data, colorImageBuffer, 3 * cameraInfo.color.height * cameraInfo.color.width * sizeof(uint8_t));

	cvtColor(rgbImage, rgbImage, CV_RGB2BGR);

I can remember that I had similar issues in the beginning (three times the same image with offset). Cannot remember the exact reason but it was something with the conversion/casting of the data I reckon.

MasWag, 2015-07-09:
@michaelwolfram 
My readme was old. Actually, libfreenect2 added Sequence support. So now my driver works with libfreenect2 master.I haven't tested it with latest version,but I tested in May.

Most of application using OpenNI2 should work with my driver. Record/play is OK but Nite did not work.
Actually I have never used @hanyazou version, so I don't know well about it. But record/play seems implemented and Nite seems to work.

rossifier, 2015-07-10:
@michaelwolfram Yes exactly, everything should work exactly as you use an xtion device. At the end you will write openni programs but dedicated to work with a kinect2 device. Normally you can still use libfreenect2 also. But the beauty of using MasWag's driver or henyazou's is that you will have a program that works with either an xtion device or a kinect2 device. You just need to switch between devices without changing your code.
Thanks for the hints with color image. I will try this out and I will tell you :)
Cheers,

rossifier, 2015-07-10:
@michaelwolfram It doesnt work for me. My color image is still duplicated three times. Did someone have the same bug?

rossifier, 2015-07-10:
@MasWag I found the bug. Actually it is in the class LF2ImageStream of the driver. You assume the data is of type RGB888. But actually the frame has 4 channels as the Protonect.cpp example shows. So I modified the size of the frameout in BuildFrame recompiled the library. And now it works. :)

hanyazou, 2015-07-10:
@rossifier libfreenect2 was changed to handle 4 channels last month. My driver was fixed in 7753bf7 already. I can see that OpenNI2's NiViewer works well on my mac.

MasWag, 2015-07-10:
@rossifier Thank you. I'm glad if you fork and commit. If you make pull request, I'll merge it.

rossifier, 2015-07-15:
@MasWag when I fork from here :https://github.com/MasWag/OpenNI2/tree/libfreenect2/Source/Drivers/Libfreenect2 I don't see the folder Libfreenect2 in my repository.

MasWag, 2015-07-15:
@rossifier 
Is your branch libfreenect2? If not , you cannot see the folder.

rossifier, 2015-07-16:
@MasWag Done :)

rossifier, 2015-07-17:
@hanyazou   I just tested your driver. It works perfectly for me also on my Ubunty 14.04 system. I just needed to add this line in my CMakeLists.txt file so that it can add C++11 features during compilation:
set(CMAKE_CXX_FLAGS "-std=c++0x ${CMAKE_CXX_FLAGS} -g -ftest-coverage -fprofile-arcs")
Thanks for your work.
Cheers,

[36 more replies]

closed 2016-01-21 by xlz