OpenKinect/libfreenect2 #244: Bug : Lack of binary release

elliotwoods, 2015-05-26:
This kind of library could really do with a binary release linked from the Readme for each platform.
Here's my build from Windows if it's useful for anyone:

https://drive.google.com/open?id=0B_S1z3_AWMRSR3A0TDBTazlmUUU&authuser=0

floe, 2015-05-29:
Thanks! This is certainly also on the list for the 0.1 release.

xlz, 2016-01-04:
A Windows binary release would include these files:
```
bin/
    turbojpeg.dll (BSD)
    glfw3.dll (zlib/libpng license, BSD-like)
    libusb-1.0.dll (modified, LGPL)
    Protonect.exe
    libfreenect2.dll (vs2013 version)
examples/
    HOWTO.txt
    CMakeLists.txt
    Protonect.cpp
    viewer.cpp
    viewer.h
lib/vs2013/
    libfreenect2.dll
    libfreenect2.lib
    libfreenect2.pdb
    libfreenect2-openni2.dll
    libfreenect2-openni2.pdb
lib/vs2015/
    ....
include/libfreenect2/
INSTALL.txt - Instructions about Zadig installer, VC runtime installer
LICENSES.txt
```

xlz, 2016-01-08:
There is a blocking problem though. Should libfreenect2 binary depends on OpenCL or not?
* If yes, users without OpenCL cannot use the binary's basic OpenGL feature.
* If no, users with OpenCl cannot use the binary's OpenCL feature.
I'm not very keen on making multiple release variants regarding feature dependencies.

xlz, 2016-01-09:
Here is a binary release for testing.

https://github.com/xlz/libfreenect2/releases/tag/0.1.0-testing

floe, 2016-01-18:
My suggestion would be to leave OpenCL out of the binary release. You will need to install some OpenCL SDK anyway if you want to use it, OpenGL on the other hand is more or less standard.

xlz, 2016-01-18:
That's what I'm doing.

xlz, 2016-01-21:
Files are to be attached to the 0.1 release.

closed 2016-01-21 by xlz