OpenKinect/libfreenect2 #245: Make error compiling with ARM. (Odroid XU3)

manusus, 2015-05-28:
I'm trying to compile libfreenect2 on UBUNTU 14.04 in an ARM platform (ARM-Cortex™-A15 and Cortex™-A7). All works fine until arrive to make when I get the following error:

[ 95%] Building C object CMakeFiles/freenect2.dir/src/flextGL.c.o
[ 97%] Building CXX object CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o
Linking CXX shared library lib/libfreenect2.so
/usr/bin/ld.bfd.real: /usr/lib/gcc/arm-linux-gnueabihf/4.8/../../../arm-linux-gnueabihf/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/arm-linux-gnueabihf/4.8/../../../arm-linux-gnueabihf/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libfreenect2.so] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

Any idea how to fix this error? The error says that "recompile with -fPIC". I've opened Makefile but I don't know where to put exactly this flag.

Thank in advance,
Manu.

wiedemeyer, 2015-05-28:
Hi,
you either have to recompile libturbojpeg with the -fPIC option, or install a libturbojpeg.so. On other 14.04 platforms, there is a symlink from libturbojpeg.so.1 to libturbojpeg.so missing in `/usr/lib`. Maybe that's also the case for the ARM platform, if so, you could simple add that symlink.

manusus, 2015-05-28:
Hi,
Thank you for your quick response. I have libturbojpeg already installed:

Building dependency tree       
Reading state information... Done
libturbojpeg is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.

I've tried to modify CMakelists.txt commenting the libturbojpeg from build in the following lines:

SET(LIBRARIES
  usb-1.0
  ${OpenCV_LIBS}
  #turbojpeg
  ${LIBFREENECT2_THREADING_LIBRARIES})

Then I've made:
 #cmake CMakelists.txt
 #make 

and now I get this new error:

[ 97%] Built target freenect2
Scanning dependencies of target Protonect
[100%] Building CXX object CMakeFiles/Protonect.dir/Protonect.cpp.o
Linking CXX executable bin/Protonect
lib/libfreenect2.so: undefined reference to `tjDestroy'
lib/libfreenect2.so: undefined reference to `tjInitDecompress'
lib/libfreenect2.so: undefined reference to `tjDecompress2'
lib/libfreenect2.so: undefined reference to `tjGetErrorStr'
collect2: error: ld returned 1 exit status
make[2]: *** [bin/Protonect] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

On the other hand, in case of recompile the library with -fPic option, which is the correct Makefile to add this option?

Thank you,
Manu.

xlz, 2015-05-29:
Duplicate of #36. It was asking for fPIC because it first tried dynamic linking with libturbojpeg.so but did not find it, then it tried static linking with libturbojpeg.a but the static one is not built with fPIC.

What you need to do is just follow the instructions in #36.

closed 2015-05-29 by floe

manusus, 2015-05-29:
Ok! Thanks, changing SHARED for STATIC library "make" works.

CrazyPopLin, 2015-11-03:
Hi @manusus , could you tell me how do you made configureation of OpenCL dependency in odroid? Thanks so much.