OpenKinect/libfreenect2 #246: [UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER when running ./bin/Protonect

tenderlygloriousgentleman, 2015-05-29:
Hi,

I'm looking for help in running the bin/Protonect program successfully.

My machine is a HP-EliteBook-8540w, and it has an NVIDIA GT216LM [Quadro FX 880M] graphics card.

I did not do anything to enable OpenCL processing.

I think I am using 64-bit Ubuntu 14.14.1. This is the output of `uname -a`:
amber@amber-HP-EliteBook-8540w:~$ uname -a
Linux amber-HP-EliteBook-8540w 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri May 8 09:43:57 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

I managed to successfully build the libfreenect2 by following the guide on (https://github.com/wiedemeyer/libfreenect2).

I made sure to add permissions to access the kinect: (https://github.com/code-iai/iai_kinect2#permissions-to-access-the-kinect-one).

I also made sure to make the libusb modifications at (https://github.com/code-iai/iai_kinect2#modifications-to-upstream-libfreenect2).

When I run the program with `sudo ./bin/Protonect`, I get this libusb error: [UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER 

(The entire text from the console ouput is pasted at the bottom.)

Could this be a problem with my usb port? lsusb shows that there is a USB 3.0 port:

Bus 002 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 023: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 022: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 012: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 04f2:b15e Chicony Electronics Co., Ltd 
Bus 001 Device 004: ID 138a:0007 Validity Sensors, Inc. VFS451 Fingerprint Reader
Bus 001 Device 003: ID 03f0:231d Hewlett-Packard 4 GB Flash Drive
Bus 001 Device 002: ID 8087:0020 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

///////////

amber@amber-HP-EliteBook-8540w:~/src/libfreenect2/examples/protonect$ sudo ./bin/Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:11 with serial 014393141234
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 07 0a 07 47 4b 53 39 32 34 2e 31    31332e2e2e2e2e2e474b533932342e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[UsbControl::setIrInterfaceState] failed! libusb error -99: LIBUSB_ERROR_OTHER
[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[Freenect2DeviceImpl] started
device serial: 014393141234
device firmware: 4.1.3911.0.7
[TransferPool::submit] failed to submit transfer

floe, 2015-05-29:
What type of XHCI controller (as indicated by `lspci`) are you using?

tenderlygloriousgentleman, 2015-05-29:
Here is  the output to lspci:

amber@amber-HP-EliteBook-8540w:~$ sudo lspci
[sudo] password for amber: 
00:00.0 Host bridge: Intel Corporation Core Processor DMI (rev 11)
00:03.0 PCI bridge: Intel Corporation Core Processor PCI Express Root Port 1 (rev 11)
00:08.0 System peripheral: Intel Corporation Core Processor System Management Registers (rev 11)
00:08.1 System peripheral: Intel Corporation Core Processor Semaphore and Scratchpad Registers (rev 11)
00:08.2 System peripheral: Intel Corporation Core Processor System Control and Status Registers (rev 11)
00:08.3 System peripheral: Intel Corporation Core Processor Miscellaneous Registers (rev 11)
00:10.0 System peripheral: Intel Corporation Core Processor QPI Link (rev 11)
00:10.1 System peripheral: Intel Corporation Core Processor QPI Routing and Protocol Registers (rev 11)
00:16.0 Communication controller: Intel Corporation 5 Series/3400 Series Chipset HECI Controller (rev 06)
00:16.3 Serial controller: Intel Corporation 5 Series/3400 Series Chipset KT Controller (rev 06)
00:19.0 Ethernet controller: Intel Corporation 82577LM Gigabit Network Connection (rev 06)
00:1a.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 06)
00:1b.0 Audio device: Intel Corporation 5 Series/3400 Series Chipset High Definition Audio (rev 06)
00:1c.0 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 (rev 06)
00:1c.1 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 2 (rev 06)
00:1c.3 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 4 (rev 06)
00:1c.7 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 8 (rev 06)
00:1d.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 06)
00:1e.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev a6)
00:1f.0 ISA bridge: Intel Corporation Mobile 5 Series Chipset LPC Interface Controller (rev 06)
00:1f.2 SATA controller: Intel Corporation 5 Series/3400 Series Chipset 6 port SATA AHCI Controller (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GT216GLM [Quadro FX 880M] (rev a2)
01:00.1 Audio device: NVIDIA Corporation GT216 HDMI Audio Controller (rev a1)
44:00.0 Network controller: Intel Corporation Centrino Ultimate-N 6300 (rev 35)
45:00.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 03)
46:06.0 FireWire (IEEE 1394): Ricoh Co Ltd R5C832 IEEE 1394 Controller (rev 06)
46:06.1 SD Host controller: Ricoh Co Ltd R5C822 SD/SDIO/MMC/MS/MSPro Host Adapter (rev 25)
46:06.2 System peripheral: Ricoh Co Ltd R5C592 Memory Stick Bus Host Adapter (rev 14)
46:06.3 System peripheral: Ricoh Co Ltd xD-Picture Card Controller (rev 14)
46:06.4 CardBus bridge: Ricoh Co Ltd RL5c476 II (rev bb)
ff:00.0 Host bridge: Intel Corporation Core Processor QuickPath Architecture Generic Non-Core Registers (rev 04)
ff:00.1 Host bridge: Intel Corporation Core Processor QuickPath Architecture System Address Decoder (rev 04)
ff:02.0 Host bridge: Intel Corporation Core Processor QPI Link 0 (rev 04)
ff:02.1 Host bridge: Intel Corporation Core Processor QPI Physical 0 (rev 04)
ff:03.0 Host bridge: Intel Corporation Core Processor Integrated Memory Controller (rev 04)
ff:03.1 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Target Address Decoder (rev 04)
ff:03.4 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Test Registers (rev 04)
ff:04.0 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 0 Control Registers (rev 04)
ff:04.1 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 0 Address Registers (rev 04)
ff:04.2 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 0 Rank Registers (rev 04)
ff:04.3 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 0 Thermal Control Registers (rev 04)
ff:05.0 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 1 Control Registers (rev 04)
ff:05.1 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 1 Address Registers (rev 04)
ff:05.2 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 1 Rank Registers (rev 04)
ff:05.3 Host bridge: Intel Corporation Core Processor Integrated Memory Controller Channel 1 Thermal Control Registers (rev 04)
amber@amber-HP-EliteBook-8540w:~$

xlz, 2015-05-30:
Have you `sudo sh -c 'for i in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$i; done'` after connecting Kinect2 before running Protonect?

xlz, 2015-05-30:
Also, relevant `dmesg`?

xlz, 2015-05-30:
Similar to #123 #126. They problem was insufficient PCIe support for USB3. But please provide the information anyway.

tenderlygloriousgentleman, 2015-06-01:
xlz, I tried your suggested command, but the command did not fix the problem.

I pasted the relevant dmesg messages at the end of this comment per your request.

I took a look at issues 123 and 126, and the problem does look similar to mine. But I want to make sure I'm not overlooking something.

Page 6 of this document (http://www8.hp.com/h20195/v2/GetDocument.aspx?docname=c04141335) says that my machine uses a Mobile Intel QM57 Express Chipset.

This page (http://www.intel.com/Products/Notebook/Chipsets/QM57/qm57-overview.htm) for the chipset lists PCI Express* 2.0 interface as one of the features.

The conclusion that issue #123 and #126 came up with was that PCIe 1.1 was not fast enough, but in my case, isn't my chipset using PCIe 2.0?

///////////

[ 1282.824189] usb 4-1: new SuperSpeed USB device number 2 using xhci_hcd
[ 1282.844150] usb 4-1: New USB device found, idVendor=045e, idProduct=02d9
[ 1282.844158] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1282.844163] usb 4-1: Product: NuiSensor Adaptor      
[ 1282.844168] usb 4-1: Manufacturer: Microsoft Corporation  
[ 1282.845073] hub 4-1:1.0: USB hub found
[ 1282.845535] hub 4-1:1.0: 1 port detected
[ 1283.179667] usb 3-1: new high-speed USB device number 3 using xhci_hcd
[ 1283.311737] usb 3-1: New USB device found, idVendor=045e, idProduct=02d9
[ 1283.311745] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1283.311751] usb 3-1: Product: NuiSensor Adaptor      
[ 1283.311755] usb 3-1: Manufacturer: Microsoft Corporation  
[ 1283.312809] hub 3-1:1.0: USB hub found
[ 1283.313188] hub 3-1:1.0: 1 port detected
[ 1284.366531] usb 4-1.1: new SuperSpeed USB device number 3 using xhci_hcd
[ 1284.386143] usb 4-1.1: New USB device found, idVendor=045e, idProduct=02c4
[ 1284.386151] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 1284.386157] usb 4-1.1: Product: Xbox NUI Sensor
[ 1284.386161] usb 4-1.1: Manufacturer: Microsoft
[ 1284.386165] usb 4-1.1: SerialNumber: 072138799988
[ 1284.438600] usbcore: registered new interface driver snd-usb-audio
[ 1333.545871] usb 4-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[ 1333.562472] xhci_hcd 0000:45:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880408c14000
[ 1333.562480] xhci_hcd 0000:45:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880408c14048
[ 1333.562486] xhci_hcd 0000:45:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880408c140d8
[ 1333.562491] xhci_hcd 0000:45:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880408c14090
[ 1333.922639] usb 4-1.1: Not enough bandwidth for new device state.
[ 1333.922652] usb 4-1.1: Not enough bandwidth for altsetting 1

xlz, 2015-06-01:
```
[ 1333.922639] usb 4-1.1: Not enough bandwidth for new device state.
[ 1333.922652] usb 4-1.1: Not enough bandwidth for altsetting 1
```
This is the same issue with 123 and 126. Those two bugs were closed by reporters without proper resolution. I think we might come up with a workaround for this, otherwise we can also document this known issue.

jsr38, 2015-07-24:
Hi there!  We're having this exact issue too.  I haven't had a look at the code yet but any chance someone can give some guidance as to what needs to be done as I'm happy to pick this up and work on it.  I gather it involves requesting less bandwidth but any idea on practical limits or potential issues I need to be aware of?  I'll start reading the code now.

floe, 2015-08-20:
Just for the record, I was setting up libfreenect2 today on a different machine with a TS-PDU3 add-on USB3 adapter (which is known to work in general). I was getting exactly the same error while the card was plugged into a PCI-X 1.1 slot; as soon as I switched to a PCI-X 2.0 slot, everything started working. I guess this really is a hardware limitation (and until now, I didn't know that there are machines which _mix_ PCI-X 1.1 and 2.0 slots on the same board).

jsr38, 2015-09-29:
I actually went in to the USB 3.0 / Intel XCHI specs and looked at the driver code.  The insufficient bandwidth response to a USB 'configure endpoint' command is made by the card firmware which presumably has rightly or wrongly presumed insufficient bandwidth due to the slot/bus.  I am going to go the new motherboard route for now as I think attempting to modify the card firmware is likely to be difficult if at all possible given limited resources :-)

xlz, 2015-10-24:
Indeed the bandwidth reservation is done by the device's firmware with configure endpoint command. According to a software bandwidth checking formula (`xhci_get_ss_bw_consumed()` in Linux kernel), Kinect2's isochronous endpoint will reserve 2.47 Gbps bandwidth. The RGB endpoint does not reserve bandwidth, but still effectively consumes 0.25 Gbps. Note that these two endpoints transfer in burst and stay idle most of the time, requiring extra margin of bandwidth.

PCIe v1.x single lane bandwidth is 2 Gbps. The isochronous endpoint will be unable to reserve required bandwidth. PCIe v2.x single lane is 4 Gbps, better. PCIe v1.x multi-lane might also work.

closed 2015-12-23 by floe