OpenKinect/libfreenect2 #247: Error running Protonect example

manusus, 2015-05-29:
Hi,

After build the library, when I try to test Protonect example I get the following error:

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:7 with serial 504550242542
[Freenect2Impl] found 1 devices
libGL error: MESA-LOADER: malformed or no PCI ID
libGL error: dlopen /usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen ${ORIGIN}/dri/exynos_dri.so failed (${ORIGIN}/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen /usr/lib/dri/exynos_dri.so failed (/usr/lib/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos

Any idea of this libGL error? I tried to reinstall MESA drivers, but it doesn't work again.

Thanks in advance,

floe, 2015-05-29:
I'm pretty sure that is not a problem with libfreenect2, but rather with your Mesa/OpenGL setup. Can you run a basic OpenGL program like glxgears?

manusus, 2015-05-29:
Yes, it seems it's not related with libfreenect. Glxgears runs the window with gears ok, but I get the same errors.... :(

libGL error: MESA-LOADER: malformed or no PCI ID
libGL error: dlopen /usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen ${ORIGIN}/dri/exynos_dri.so failed (${ORIGIN}/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen /usr/lib/dri/exynos_dri.so failed (/usr/lib/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
820 frames in 5.0 seconds = 163.922 FPS
789 frames in 5.0 seconds = 157.667 FPS
794 frames in 5.0 seconds = 158.698 FPS

Thank you for your response.

closed 2015-05-29 by floe