OpenKinect/libfreenect2 #248: No depth stream

SamTew, 2015-05-30:
When i am running kinect2_bridge, there are rgb stream. I can access rgb image but no depth image. When i try to rostopic echo /kinect2/depth/image, nothing was shown. This situation happen when i compile freenect2 with opengl and opencl are disabled.
cmake ../../examples/protonect/ -DENABLE_CXX11=ON -DENABLE_OPENCL=OFF -DENABLE_OPENGL=OFF
I would like to ask whether OpenGL is necessary to get depth data stream?

larshg, 2015-05-30:
Opengl and opencl is not necessary as there is a cpu implementation as well. What does the output in the console say?

SamTew, 2015-05-30:
default ir camera parameters: 
fx 365.861, fy 365.861, cx 254.502, cy 205.253
k1 0.0964056, k2 -0.275756, p1 0, p2 0, k3 0.0921489

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5
using sensor defaults for color intrinsic parameters.
using sensor defaults for ir intrinsic parameters.
using defaults for rotation and translation.
using defaults for depth shift.

camera parameters used:
camera matrix color:
[1081.3720703125, 0, 959.5;
  0, 1081.3720703125, 539.5;
  0, 0, 1]
distortion coefficients color:
[0, 0, 0, 0, 0]
camera matrix ir:
[365.8608093261719, 0, 254.502197265625;
  0, 365.8608093261719, 205.2525939941406;
  0, 0, 1]
distortion coefficients ir:
[0.09640563279390335, -0.2757557928562164, 0, 0, 0.09214892983436584]
rotation:
[1, 0, 0;
  0, 1, 0;
  0, 0, 1]
translation:
[-0.052;
  0;
  0]
depth shift:
0

[DepthRegistration::New] Using CPU registration method!
[DepthRegistration::New] Using CPU registration method!
[kinect2_bridge] waiting for clients to connect

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
[RgbPacketStreamParser::onDataReceived] packetsize or sequence doesn't match!
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[kinect2_bridge] depth processing: ~0Hz (infms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~57.0852Hz (17.5177ms) publishing rate: ~21.9309Hz
[TurboJpegRgbPacketProcessor] avg. time: 23.8427ms -> ~41.9416Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~56.9334Hz (17.5644ms) publishing rate: ~29.96Hz
[TurboJpegRgbPacketProcessor] avg. time: 24.5025ms -> ~40.8122Hz
[kinect2_bridge] depth processing: ~-nanHz (-nanms) publishing rate: ~0Hz
[kinect2_bridge] color processing: ~61.4914Hz (16.2624ms) publishing rate: ~29.9774Hz

SamTew, 2015-05-30:
warning: ignoring unsupported tag `MARKDOWN_SUPPORT       =' at line 253, file Doxyfile
warning: ignoring unsupported tag `AUTOLINK_SUPPORT       =' at line 260, file Doxyfile
warning: ignoring unsupported tag `EXTRACT_PACKAGE        =' at line 362, file Doxyfile
warning: ignoring unsupported tag `USE_MDFILE_AS_MAINPAGE =' at line 790, file Doxyfile
warning: ignoring unsupported tag `HTML_EXTRA_STYLESHEET  =' at line 928, file Doxyfile
warning: ignoring unsupported tag `HTML_INDEX_NUM_ENTRIES =' at line 985, file Doxyfile
warning: ignoring unsupported tag `MATHJAX_FORMAT         =' at line 1206, file Doxyfile
warning: ignoring unsupported tag `EXTERNAL_SEARCH        =' at line 1253, file Doxyfile
warning: ignoring unsupported tag `SEARCHENGINE_URL       =' at line 1261, file Doxyfile
warning: ignoring unsupported tag `SEARCHDATA_FILE        =' at line 1267, file Doxyfile
warning: ignoring unsupported tag `EXTERNAL_SEARCH_ID     =' at line 1274, file Doxyfile
warning: ignoring unsupported tag `EXTRA_SEARCH_MAPPINGS  =' at line 1283, file Doxyfile
warning: ignoring unsupported tag `UML_LIMIT_NUM_FIELDS   =' at line 1744, file Doxyfile

I got these warning during ./install_ubuntu.sh. Are these warning causing problem?

larshg, 2015-05-30:
Hmm, I don't use the bridge nor ros, so you might be better off posting your problem athttps://github.com/code-iai/iai_kinect2 or try run protonect as it's the one I'm familiar with. Maybe @wiedemeyer can help?

SamTew, 2015-05-31:
I post at both side. Haha. 
It's ok. Thanks anyway.

wiedemeyer, 2015-06-01:
I think there is a problem with the USB, because you are not receiving depth and ir images. Does it say `[TransferPool::submit] failed to submit transfer` when you start `Protonect`? is `Protonect` showing depth and ir images?

lakehanne, 2015-06-04:
Anyone know how to fix this? I keep getting errors when I run make && sudo make install after cmake

>Linking CXX executable /home/lex/libfreenect2/examples/protonect/bin/Protonect
/home/lex/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined reference to `XF86VidModeQueryExtension'
/home/lex/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined reference to `XF86VidModeGetGammaRampSize'
/home/lex/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined reference to `XF86VidModeGetGammaRamp'
/home/lex/libfreenect2/examples/protonect/lib/libfreenect2.so: undefined reference to `XF86VidModeSetGammaRamp'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/lex/libfreenect2/examples/protonect/bin/Protonect] Error 1
make[1]: *** [CMakeFiles/Protonect.dir/all] Error 2

SamTew, 2015-06-05:
sudo apt-get install cmake xorg-dev libglu1-mesa-dev
maybe u can try this but be careful whether some packages are going to remove or not.

lakehanne, 2015-06-08:
Thanks! I am using OpenGL and here is my output:

>lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:9 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[CommandTransaction::receive] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
Segmentation fault (core dumped)
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect cpu
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:9 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[CommandTransaction::receive] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
Segmentation fault (core dumped)
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ldd /usr/local/lib/libfreenect2.so | grep "libfreenect2"
	libusb-1.0.so.0 => /home/lex/Documents/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f3630eba000)
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ldd /usr/local/lib/libusb | grep "libfreenect2"
ldd: /usr/local/lib/libusb: No such file or directory
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ldd /usr/local/lib/libfreenect2.so | grep "libfreenect2"
	libusb-1.0.so.0 => /home/lex/Documents/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f42f46fe000)
lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:9 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[CommandTransaction::receive] bulk transfer failed! libusb error -1: LIBUSB_ERROR_IO
Segmentation fault (core dumped)

 Beat it!

lakehanne, 2015-06-08:
Never mind, I fixed this by installing the libusbX superspeed patch from Josh Blake's github account. 
However, I notice my depth map is very noisy compared with last weekend when I ran Protonect with the `./Protonect cpu` option.
Here is my current output from `./Protonect gl`

>lex@lex:~/Documents/libfreenect2/examples/protonect/bin$ ./Protonect gl
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 09 13 11 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 501185143042
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 1022
[OpenGLDepthPacketProcessor] avg. time: 6.26457ms -> ~159.628Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.964ms -> ~58.9483Hz
[OpenGLDepthPacketProcessor] avg. time: 5.86282ms -> ~170.566Hz
[TurboJpegRgbPacketProcessor] avg. time: 17.0024ms -> ~58.8153Hz
[OpenGLDepthPacketProcessor] avg. time: 5.87456ms -> ~170.226Hz
[TurboJpegRgbPacketProcessor] avg. time: 16.8351ms -> ~59.3996Hz
[OpenGLDepthPacketProcessor] avg. time: 5.88891ms -> ~169.811Hz
[TurboJpegRgbPacketProcessor] avg. time: 17.1733ms -> ~58.23Hz
[OpenGLDepthPacketProcessor] avg. time: 5.91201ms -> ~169.147Hz
[TurboJpegRgbPacketProcessor] avg. time: 17.0803ms -> ~58.5469Hz
[OpenGLDepthPacketProcessor] avg. time: 5.88296ms -> ~169.983Hz

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe