OpenKinect/libfreenect2 #249: LIBUSB_ERROR_ACCESS error

bmiller59, 2015-06-01:
I just followed the install instructions on OSX Yosemite and it all seemed to go fine during the build.

I think tried to run Protonect and I get this error. 

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 10 usb devices connected
[Freenect2Impl] found valid Kinect v2 @250:8 with serial 503770641942
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -3: LIBUSB_ERROR_ACCESS
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @250:8!
failure opening device!

I have tried sudoing the command and adding my user to the admin and wheel groups: didn't work. I do have a variety of other usb devices running through a usb hub plugged into the other usb port. 

Any suggestions?

gaborpapp, 2015-06-01:
What do you see if you only connect the Kinect2 and remove all other usb devices?

bmiller59, 2015-06-03:
With all of the other usb devices disconnected I get the same error message,

randomize, 2015-08-06:
Also getting this error message. OSX Yosemite (14.3.0 Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64 x86_64) and current master branch of libfreenect2 (2a3a948177404a0dca6ed26f71d88abc415e2188)

ahundt, 2015-08-07:
I may have encountered this, and it may be documented in another issue. 

working:
1) Protonect cv

working with gfxCardStatus set to Integrated Only:
2) Protonect 

not working with discrete graphics:
3) Protonect 

Try:

 - unplug power from kinect
 - plug kinect into usb3 port
 - plug power into kinect
 - check the app "System Information" for SuperSpeed usb 3.0, not Hi-speed

Relevant issues:
#78 #31 #310 

If this fixes your problem, these details really need to be added to the README.md

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe

bmiller59, 2016-03-29:
I just downloaded and built the latest version and I am still getting the same error. I will reopen as suggested.

bjeanes, 2021-11-22:
I also get this. Increasing logging I see:

```
❯ env LIBUSB_DEBUG=3 ./bin/Protonect cl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Apple M1 Max (GPU)[Apple]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Apple M1 Max (GPU)[Apple]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 2 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @0:2 with serial 128533634047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_ACCESS Access denied (insufficient permissions). Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
libusb: warning [darwin_release_interface] USBInterfaceClose: no connection to an IOService
libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @0:2
failure opening device!
```

mr-z-ro, 2025-02-09:
@bjeanes did you solve this? Experiencing the same issue.