OpenKinect/libfreenect2 #252: Windows USB trouble with final device, but not preview one

HenningJ, 2015-06-02:
I guess this is not a libfreenect2 bug, but maybe you guys can help anyways...

I'm trying to get the Kinect 2 to work on Windows 7 and I'm having a little trouble...
When I connect my preview Kinect 2 (045e:02c4) to the USB3.0 port (NEC/Renesas USB controller), it shows up as superspeed device and everything works fine.
But when I connect the final hardware Kinect 2 (045e:02d8) to the same USB port, it shows up as a USB 2 (Highspeed) device, and of course won't work with libfreenect2:

```
[UsbControl::enablePowerStates(U1)] failed! libusb error -9: LIBUSB_ERROR_PIPE
```

I don't think it's a libusb(x) driver problem, because it shows up as some kind of simulated USB 2.0 device in the device-manager even before I do the whole [zadig driver replacement dance](https://github.com/OpenKinect/libfreenect2#windows):

[image]

After replacing the driver it's pretty much the same:

[image]

The preview device shows up correctly as USB 3.0:

[image]

That leads me to believe this is some kind of hardware/firmware problem. Is there a firmware update which might help?
Firmware updates can only be done using the official SDK (and therefore Windows 8), right?

floe, 2015-06-02:
Yes, unfortunately, firmware updates only work with the official SDK. I'm not sure what the most recent firmware version is, what does Protonect output as FW version when you run it on the working device?

HenningJ, 2015-06-02:
For the working (preview) device, protonect says:
```
device firmware: 2.3.3913.0.7
```

HenningJ, 2015-06-03:
Hmmm...interesting. I just tried to use the newer device with the older connector box thingy...and it works. It has a slightly lesser firmware version, but seems to work fine now:
```
device serial: 504671542542
device firmware: 2.3.3912.0.7
```
Interestingly, when I plug the older device into the newer connector box, it also works.
So, the only thing that doesn't work is the newer device with the newer connector box. Weird.

But, since I have two working sensors now, I'm closing the issue. :)

closed 2015-06-03 by HenningJ

HenningJ, 2015-06-29:
Btw. after updating the firmware (it's version 4.1.3911.0.7 now), both devices work fine in any combination with both of the connector boxes.

floe, 2015-06-29:
What do you need to do to update the firmware? I'm wondering if it would be possible to capture the USB traffic during the update...

HenningJ, 2015-06-29:
When you plug the Kinect into a Windows 8 PC, once the drivers are installed, the firmware is updated automatically. This is mentioned on the ["installation guide"](https://msdn.microsoft.com/en-us/library/dn782035.aspx#Installation):
> On first plugin, the firmware on the device will be updated. This may result in device enumeration happening several times in the first minute. 

So I guess you'd just need a Kinect with an outdated firmware and Windows 8...and good timing.