OpenKinect/libfreenect2 #254: Segmentation Error after libGL Error

manusus, 2015-06-02:
Hi,

I'm trying to test Protonect example but I get the following error:

odroid@odroid:~/libfreenect2/examples/protonect/bin$ ./Protonect 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 504550242542
[Freenect2Impl] found 1 devices
libGL error: MESA-LOADER: malformed or no PCI ID
libGL error: dlopen /usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen ${ORIGIN}/dri/exynos_dri.so failed (${ORIGIN}/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: dlopen /usr/lib/dri/exynos_dri.so failed (/usr/lib/dri/exynos_dri.so: cannot open shared object file: No such file or directory)
libGL error: unable to load driver: exynos_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: exynos
Segmentation fault

I've been asking in Odroid Forum, and it seems that this error can be ignored. I've tried to load OpenGL examples as glxgears and works fine, but showing this error in terminal. I think the problem is there is no support to libGL in Odroid devices... :(
There is another way to load kinect2 without OpenGL? My idea is to use after that, OpenCV in order to work with depth images. I only need depth images.
Some idea?

Thanks in advance,

floe, 2015-06-02:
You can try to run `Protonect cpu` or `Protonect cl` and see if any of those work (as also mentioned in the FAQ). Otherwise, you could try to get a backtrace from that segmentation fault.

manusus, 2015-06-02:
Thank you, I've tried to run the program with cl option and throw the same error, but running the program with cpu option, it seems that works, but no image is shown on the windows registered, depth and ir.

The terminal shows the following:

[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 504550242542
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 06 0f 06 47 4d 46 31 33 33 2e 31    31332e2e2e2e2e2e474d463133332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started
device serial: 504550242542
device firmware: 4.3.3912.0.7
[TurboJpegRgbPacketProcessor] avg. time: 59.3087ms -> ~16.8609Hz
[Freenect2DeviceImpl] started
device serial: 504550242542
device firmware: 4.3.3912.0.7
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[DepthPacketStreamParser::onDataReceived] subpacket too large
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] not all subsequences received 576
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[DepthPacketStreamParser::onDataReceived] skipping depth packet
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[Freenect2DeviceImpl] stopped
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] releasing usb interfaces...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] already closed, doing nothing

Thanks!

xlz, 2015-06-03:
CPU depth processor is extremely demanding on the CPU and probably prevents USB isochronous transfers from being properly processed on slow CPUs. I think Odroid hardware spec is insufficient for even 1Hz.

manusus, 2015-06-03:
Odroid XU3 is quite powerful, but maybe not enough. These are the XU3 specifications:

* Samsung Exynos5422 Cortex™-A15 2.0Ghz quad core and Cortex™-A7 quad core CPUs
* Mali-T628 MP6(OpenGL ES 3.0/2.0/1.1 and OpenCL 1.1 Full profile)
* 2Gbyte LPDDR3 RAM at 933MHz (14.9GB/s memory bandwidth) PoP stacked

I'm going to try to use OpenCL.

Thanks,

manusus, 2015-06-03:
With OpenCL I get this message:

odroid@odroid:~/libfreenect2/examples/protonect/bin$ sudo ./Protonect cl
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 504550242542
[Freenect2Impl] found 1 devices
OpenCL pipeline is not supported!

xlz, 2015-06-03:
Mali GPU 100Gflops is estimated to output 20Hz. If you can make Mali OpenCL  SDK work, sure.

xlz, 2015-06-03:
You need to enable OpenCL during build.

manusus, 2015-06-03:
Ok. But it seems is already enable. Do you refer to enable/disable options in CMakeFile.txt?
On the other hand, and trying to find some incompatibilities, I have installed OpenCV too, it could be a problem? I see in CMakeFile.txt references about OpenCV. What I should install before? OpenCV or libFreenect2?

These are the first lines of CMakeList.txt where it seems OpenCL support is already enable:  

CMAKE_MINIMUM_REQUIRED(VERSION 2.6)

PROJECT(libfreenect2)
SET(CMAKE_BUILD_TYPE RelWithDebInfo)

SET(MY_DIR ${libfreenect2_SOURCE_DIR})

OPTION(ENABLE_CXX11 "Enable C++11 support" OFF)
OPTION(ENABLE_OPENCL "Enable OpenCL support" ON)
OPTION(ENABLE_OPENGL "Enable OpenGL support" ON)

Thank you very much.

xlz, 2015-06-03:
Does Cmake output "Found OpenCL"?

manusus, 2015-06-04:
Hi,

Thank you very much xlz. I've re installed all the operating system again, and the libraries step by step. On the CMake point, and asking your question about output of CMake, It doesn't arrive to "Found OpenCL" because a previous error:

odroid@odroid:~/libfreenect2/examples/protonect$ cmake CMakeLists.txt
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Failed
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb'
--   found libusb, version 0.1.12
-- checking for module 'opencv'
--   found opencv, version 2.4.9
-- TurboJPEG_INCLUDE_DIR = /usr/include
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- TurboJPEG_LIBRARY = /usr/lib/arm-linux-gnueabihf/libturbojpeg.so
-- checking for module 'glfw3'
--   package 'glfw3' not found
CMake Error at CMakeLists.txt:135 (if):
  if given arguments:

    "STREQUAL" "FALSE"
  Unknown arguments specified
-- Configuring incomplete, errors occurred!
See also "/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeError.log".

And the CMakeError:

odroid@odroid:~/libfreenect2/examples/protonect/CMakeFiles$ cat CMakeError.log 
Performing C++ SOURCE FILE Test LIBFREENECT2_THREADING_STDLIB failed with the following output:
Change Dir: /home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec545066115/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec545066115.dir/build.make CMakeFiles/cmTryCompileExec545066115.dir/build
make[1]: Entering directory `/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/CMakeFiles 1
Building CXX object CMakeFiles/cmTryCompileExec545066115.dir/src.cxx.o
/usr/bin/c++    -DLIBFREENECT2_THREADING_STDLIB   -o CMakeFiles/cmTryCompileExec545066115.dir/src.cxx.o -c /home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx
In file included from /usr/include/c++/4.8/thread:35:0,
                 from /home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:2:
/usr/include/c++/4.8/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx: In function ‘int main(int, char**)’:
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:8:3: error: ‘thread’ is not a member of ‘std’
   std::thread thread;
   ^
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:8:15: error: expected ‘;’ before ‘thread’
   std::thread thread;
               ^
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:9:3: error: ‘mutex’ is not a member of ‘std’
   std::mutex mutex;
   ^
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:9:14: error: expected ‘;’ before ‘mutex’
   std::mutex mutex;
              ^
/home/odroid/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/src.cxx:10:3: error: ‘lock_guard’ is not a member of ‘std’
   std::lock_guard<std::mutex> lock_guard(mutex);
[... 33 lines cut ...]
make: *** [cmTryCompileExec545066115/fast] Error 2

Source file was:

#include <thread>
#include <mutex>
#include <condition_variable>
#include <chrono>

int main(int argc, char** argv) {
  std::thread thread;
  std::mutex mutex;
  std::lock_guard<std::mutex> lock_guard(mutex);
  std::unique_lock<std::mutex> unique_lock(mutex);
  std::condition_variable condition_variable;
  thread_local int i;
  
  return 0;
}

###########################################################################

Any idea how to fix this error?

Thanks in advance!

larshg, 2015-06-04:
Its a bug due to that pkg-config doesn't set the variable ${GLFW_FOUND} properly - you can try just to set the condition to false at that line ie. 135 in the cmake list - this will make it build the local glfw.

or you can try install GLFW3 using the package manager ? I guess you are on linux - else maybe @goldhoorn can tell us what to do :)

manusus, 2015-06-04:
Thanks for the response. Yes, it's a Lubuntu. Regarding the CMake bug, the problem is that it's already set to false :( Maybe, I should not enable OpenGL option, because ODROID XU3 it's only compatible with OpenGL ES and OpenCL. My idea is to use OpenCL.

IF(ENABLE_OPENGL)
  auto_detect_lib(GLFW3)  
  #Assuming local build witout global deps, ugly to to this automatic but to keep backward compatibiliy
  if(${GLFW3_FOUND} STREQUAL "FALSE")

manusus, 2015-06-08:
I've installed GLFW3 library and OpenCL and now CMake output arrives to "Could not find OpenCL":
It doesn't find include paths... 

odroid@odroid:~/libfreenect2/examples/protonect$ sudo cmake CMakeLists.txt
-- using tinythread as threading library
-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS) 
-- Using X11 for window creation
-- Using GLX for context creation
-- Configuring done

floe, 2015-06-08:
From FindOpenCL.cmake:

    # OpenCL_INCPATH    - Include path (e.g. OpenCL_INCPATH=/opt/cuda/4.0/cuda/include)
    # OpenCL_LIBPATH    - Library path (e.h. OpenCL_LIBPATH=/usr/lib64/nvidia)

You could try setting these on the cmake commandline to the appropriate paths for Exynos and try again?

xlz, 2015-06-08:
Generally you should install opencl-headers for standard headers, and configure your vendor specific drivers with /etc/OpenCL/... something. And test compile it with a clGetDeviceInfo.cpp.

manusus, 2015-06-10:
Thank you for responses,

Floe, do you refer to add the include and library path in CMakeLists.txt? I have the paths in /usr/lib and /usr/include/CL.
Where I should modify this paths (OPENCL_INCLUDE_DIRS and OPENCL_LIBRARIES) in order to set the paths?
Sorry, but I'm new with cmake.... :(

This a piece of CMakeLists.txt:

IF(ENABLE_OPENCL AND OPENCL_FOUND)
  SET(LIBFREENECT2_WITH_OPENCL_SUPPORT 1)
  INCLUDE_DIRECTORIES(${OPENCL_INCLUDE_DIRS})  
  LIST(APPEND SOURCES
    src/opencl_depth_packet_processor.cpp
  )

  LIST(APPEND LIBRARIES
    ${OPENCL_LIBRARIES}
  )
  
  LIST(APPEND RESOURCES
    src/opencl_depth_packet_processor.cl
  )
ENDIF(ENABLE_OPENCL AND OPENCL_FOUND)

thank you in advance,

floe, 2015-06-10:
I think you have an older revision, can you pull the most recent master branch and re-try?

manusus, 2015-06-10:
Ok, I have now the latest version but I get the same message. The point is how to set OpenCL libraries in order to libfreenect find them.

I've followed these blogs to install OpenCL on for Mali GPU (Odroid U3):
http://www.soa-world.de/echelon/2014/10/embedded-opencl-hardware.html
http://granolamatt.com/working/2015/02/configure-opencl-on-odroid-xu3/

This is the new output of Cmake:
 
 Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found OpenGL: /usr/lib/arm-linux-gnueabihf/libGL.so  
-- Could NOT find OpenCL (missing:  OPENCL_INCLUDE_DIRS) 
Linking with these libraries: opencv_viz;opencv_videostab;opencv_video;opencv_ts;opencv_superres;opencv_stitching;opencv_photo;opencv_ocl;opencv_objdetect;opencv_nonfree;opencv_ml;opencv_legacy;opencv_imgproc;opencv_highgui;opencv_gpu;opencv_flann;opencv_features2d;opencv_core;opencv_contrib;opencv_calib3d;opencv_viz;opencv_videostab;opencv_video;opencv_ts;opencv_superres;opencv_stitching;opencv_photo;opencv_ocl;opencv_objdetect;opencv_nonfree;opencv_ml;opencv_legacy;opencv_imgproc;opencv_highgui;opencv_gpu;opencv_flann;opencv_features2d;opencv_core;opencv_contrib;opencv_calib3d;usb-1.0;/usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0;pthread;glfw3;/usr/lib/arm-linux-gnueabihf/libGL.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/libfreenect2/examples/protonect

Thanks

xlz, 2015-06-11:
Have you installed `opencl-headers`?

manusus, 2015-06-12:
Hi,

Yes, I have installed them on /usr/include/CL

odroid@odroid:~/libfreenect2/examples/protonect$ ls -l /usr/include/CL
total 84
-rw-r--r-- 1 root root 12294 Apr 13  2014 cl_ext.h
-rw-r--r-- 1 root root 12294 Apr 13  2014 cl_ext.h.1
-rw-r--r-- 1 root root  6715 Apr 13  2014 cl_gl.h
-rw-r--r-- 1 root root 38036 Apr 13  2014 cl_platform.h
-rw-r--r-- 1 root root  1754 Apr 13  2014 opencl.h

But it seems that Cmake doesn't find the path to these headers. Where I could set them?

Thanks.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe