OpenKinect/libfreenect2 #258: libusb-1.0 linking breaks when libusb points to previous libusb version

Algomorph, 2015-06-04:
This should be an easy fix. I've just encountered a machine where libusb points to libusb-0.1.so.4, so -lusb produces linker errors.
The easy solution is to copy some FindLibUSB-1.0 module into an appropriate subfolder here, make sure it's added to CMake's path. For instance [the one from libfreenect](https://github.com/OpenKinect/libfreenect/blob/master/cmake_modules/Findlibusb-1.0.cmake) can be used. Then passing "libusb-1.0" in the CMakeLists.txt instead of "LibUSB" should do it.

dani-carbonell, 2015-06-04:
I think I am getting this error. 

```
dardo@dardo-GP60-2OD:~/libfreenect2/build$ cmake ../examples/protonect/ -DENABLE_CXX11=ON
-- using stdlib as threading library
CMake Error at CMakeLists.txt:9 (FIND_PACKAGE):
  By not providing "FindLibUSB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LibUSB", but
  CMake did not find one.

  Could not find a package configuration file provided by "LibUSB" with any
  of the following names:

    LibUSBConfig.cmake
    libusb-config.cmake

  Add the installation prefix of "LibUSB" to CMAKE_PREFIX_PATH or set
  "LibUSB_DIR" to a directory containing one of the above files.  If "LibUSB"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:54 (auto_detect_lib)

-- Configuring incomplete, errors occurred!
```

```
dardo@dardo-GP60-2OD:~$ locate     libusb
/lib/firmware/carl9170fw/extra/libusb-zeropacket.diff
/lib/x86_64-linux-gnu/libusb-0.1.so.4
/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4
/lib/x86_64-linux-gnu/libusb-1.0.so.0
/lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
/usr/bin/libusb-config
/usr/include/libusb-1.0
/usr/include/libusb-1.0/libusb.h
/usr/lib/libusbmuxd.so.1.0.8
/usr/lib/libusbmuxd.so.2
/usr/lib/x86_64-linux-gnu/libusb-0.1.so.4
/usr/lib/x86_64-linux-gnu/libusb-1.0.a
/usr/lib/x86_64-linux-gnu/libusb-1.0.so
/usr/lib/x86_64-linux-gnu/libusb.a
/usr/lib/x86_64-linux-gnu/libusb.so
/usr/lib/x86_64-linux-gnu/pkgconfig/libusb-1.0.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/libusb.pc
...
```

floe, 2015-06-05:
If you try branch `cmake_libusb_1.0` instead of master, does this fix things?

floe, 2015-06-07:
This should be fixed in master, if it's not, please re-open. Patched libusb should now also be found properly, even at runtime.

closed 2015-06-07 by floe

Algomorph, 2015-06-09:
Thanks. Sorry I did not have an opportunity to test the patch.