OpenKinect/libfreenect2 #273: Mac OS X, LibUSB_INCLUDE_DIR

ThomasLengeling, 2015-06-07:
I pull the latest master, everything was working in a previous commit.

any thoughts?

I get this error when doing  cmake CMakeLists.txt

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LibUSB_INCLUDE_DIR
   used as include directory in directory /Users/tom/Documents/GitHub/libfreenect2/examples/protonect
   used as include directory in directory /Users/tom/Documents/GitHub/libfreenect2/examples/protonect
   used as include directory in directory /Users/tom/Documents/GitHub/libfreenect2/examples/protonect

-- Configuring incomplete, errors occurred!
See also "/Users/tom/Documents/GitHub/libfreenect2/examples/protonect/CMakeFiles/CMakeOutput.log".
See also "/Users/tom/Documents/GitHub/libfreenect2/examples/protonect/CMakeFiles/CMakeError.log".

xlz, 2015-06-08:
Should be fixed by #274. Please `git fetch origin pull/274/head:pr274; git merge pr274` and test.

ThomasLengeling, 2015-06-08:
thanks!!, that works

closed 2015-06-08 by ThomasLengeling