OpenKinect/libfreenect2 #280: Python Bindings (not a bug)

tikiking1, 2015-06-09:
So, I wrote some quick (8 hours) Python bindings for libfreenect2, entitled [pyfreenect2](https://github.com/tikiking1/pyfreenect2). If anyone wants to help (I still have no idea what Registration is), have at it.

RyanGordon, 2015-06-10:
Good job!

Someone correct me if I'm wrong but Registration is the process of overlaying the RGB image onto the depth map?

lakehanne, 2015-06-16:
Cool. Exactly what I've been waiting for. I'll see what ConvNets can do here

xlz, 2016-01-29:
Do you have a plan to get it merged to this repo? Or do you plan to maintain it yourself (and add a mention in the README.md here)?

remexre, 2016-01-29:
Sorry, I haven't had much time to maintain this recently. In the next
couple of weeks, I'll get it "up to snuff" and then I'd be amenable to
merging.

r9y9, 2016-01-30:
FWIW, I also have a python binding (it's private for now) based on cython and also Julia binding [r9y9/Libfreenect2.jl](https://github.com/r9y9/Libfreenect2.jl) (requires development version of [julia](https://github.com/JuliaLang/julia) though). Recently I'm regularly testing with the latest libfreenect2. 

If anyone interested in python one, I will consider make it public (example code is somthing like [this](https://gist.github.com/r9y9/e928a8115fb7b6cae4ae)).

xlz, 2016-01-30:
Your python example is quite similar but has some small differences in naming and functionalities. pyfreenect2 seems to have an extra layer of python encapsulation around raw interfaces, which yours doesn't have.

I think it's best for you two to have consensus and solve the differences. Probably by providing reviews in the potential pull request. There can be only one python wrapper.

I took a brief look. "PyFreeNect2" is not consistent with the rest of the naming scheme. "PyFreenect2" or "pyfreenect2" should be used. Also, I see some underscore names. My opinion is to use CamelCase in a consistent way that matches the C++ API as much as possible to minimize confusion.

As to the Julia binding, you are welcome to make a pull request. A binding needs documentation, an example, and cross-platform test results to be merged.

floe, 2016-01-31:
For the record, the SWIG file in #427 should in the future also be able to auto-generate Python bindings. But of course, this requires working out the kinks there first, while we have working bindings here.

r9y9, 2016-02-28:
I just made my python binding public under MIT license: https://github.com/r9y9/pylibfreenect2.

In pylibfreenect2:

- All basic features (RGB, IR, depth image transfer and registration) are supported
- Recently added features (e.g. CUDA depth decoding) are not supported yet
- Tests are passing locally with the latest release (v0.1.1) and current master (https://github.com/OpenKinect/libfreenect2/commit/9c80ccd6a088a0e9e6cdafc9c837497b2434328d)
- Should be python 2.7/3.5 compatible
- No docs yet (except for brief README).
- Only tested on osx for now
- There are a few API differences in C++ and python, though I tried to minimize it (will probably be revisited)

For someone who is interested, the easiest way to use pylibfreenect2 for now is though the examples/multiframe_listener.py. 

Any comment / issue / PR is welcome.