OpenKinect/libfreenect2 #286: Error Adding Symbols: Bad Value

SebastianVarma, 2015-06-10:
@floe When setting up the library, once I enter the command "make && sudo make install", an error results.  It is as follows:

/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libfreenect2.so] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

floe, 2015-06-10:
This is very weird; this is apparently indeed a duplicate of #36, but that shouldn't happen anymore with most recent master. Only thing I can suggest at the moment is to actually remove the symlink for `libturbojpeg.so` again and give it another try...

xlz, 2015-06-11:
Can you post the log of your `cmake ..`?

(Should have `Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0`)

SebastianVarma, 2015-06-11:
@xlz Below are the CMake Error and Output logs
Output Log
The system is: Linux - 3.16.0-30-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/2.8.12.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/2.8.12.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec1360601942/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1360601942.dir/build.make CMakeFiles/cmTryCompileExec1360601942.dir/build
make[1]: Entering directory `/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1360601942.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec1360601942.dir/testCCompiler.c.o   -c /home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTryCompileExec1360601942
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1360601942.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec1360601942.dir/testCCompiler.c.o  -o cmTryCompileExec1360601942 -rdynamic 
make[1]: Leaving directory `/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec1471148617/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1471148617.dir/build.make CMakeFiles/cmTryCompileExec1471148617.dir/build
make[1]: Entering directory `/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1471148617.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec1471148617.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-2.8/Modules/CMakeCCompilerABI.c
Linking C executable cmTryCompileExec1471148617
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1471148617.dir/link.txt --verbose=1
/usr/bin/cc     -v CMakeFiles/cmTryCompileExec1471148617.dir/CMakeCCompilerABI.c.o  -o cmTryCompileExec1471148617 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.2-19ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) 
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/4.8/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTryCompileExec1471148617' '-rdynamic' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/4.8/collect2 --sysroot=/ --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -z relro -o cmTryCompileExec1471148617 /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/crt1.o /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/4.8/crtbegin.o -L/usr/lib/gcc/x86_64-linux-gnu/4.8 -L/usr/lib/g [... cut ...]
make[1]: Leaving directory `/home/sebastian/libfreenect2/examples/protonect/CMakeFiles/CMakeTmp'
make[1]: *** [CMakeFiles/cmTryCompileExec3076327744.dir/src.cxx.o] Error 1
make: *** [cmTryCompileExec3076327744/fast] Error 2

Source file was:

#include <thread>
#include <mutex>
#include <condition_variable>
#include <chrono>

int main(int argc, char** argv) {
  std::thread thread;
  std::mutex mutex;
  std::lock_guard<std::mutex> lock_guard(mutex);
  std::unique_lock<std::mutex> unique_lock(mutex);
  std::condition_variable condition_variable;
  thread_local int i;
  
  return 0;
}

xlz, 2015-06-11:
Please provide the *console output* log of `cmake ..`, not the generated log files.

SebastianVarma, 2015-06-11:
@xlz Ok.  Here is it is.

[  4%] Built target generate_resources_tool
Linking CXX shared library lib/libfreenect2.so
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libturbojpeg.a(libturbojpeg_la-turbojpeg.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libturbojpeg.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libfreenect2.so] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xlz, 2015-06-11:
No, this is the console output of `make`. Please provide the console output of `cmake ..` (after you `rm -rf` the build directory).

SebastianVarma, 2015-06-11:
Something really weird just happened.  I decided to start over the installation process since I was a little confused and it actually worked.  I think the issue could have been that I ran cmake CMakeLists.txt too many times (the first time I tried running it I got an error saying that I didn't have a dependency).  Thank you for your help.

closed 2015-06-11 by SebastianVarma

DozenDozen, 2015-11-24:
@SebastianVarma @floe I've had the exactly same problem. I tried to debug using cmake-gui and found out surprisingly that project generated by cmake-gui had no problem. This is really weird.