OpenKinect/libfreenect2 #287: Libfreenect2 Documentation

jeppewalther, 2015-06-11:
Hi
Do there exist some form of documentation on Libfreenect2 internals other that the source code itself.

SebastianVarma, 2015-06-11:
Also, are there any differences between the libfreenect and libfreenect2 api's?

floe, 2015-06-11:
@jeppewalther I guess the closest thing we have to documentation is `protonect.cpp`, unfortunately.
@SebastianVarma Biggest difference: libfreenect is mostly a synchronous C API with a C++ wrapper, libfreenect2 is pure C++ and asynchronous with internal threading.

SebastianVarma, 2015-06-11:
@floe Are the function and variable names the same?

floe, 2015-06-11:
No - it's a quite different type of API, don't think it's comparable.

SebastianVarma, 2015-06-11:
@floe Is there any sort of guide or tutorial besides the documentation (protonect.cpp)?  Could one learn the library using libfreenect2/include/libfreenect2.h?

jeppewalther, 2015-06-12:
@floe Thanks for the info. BTW The documentation i was querying for was not a user guide. But documentation on how you have implemented the driver. My guess is that there should be a paper worth publishing with this information.

floe, 2015-06-12:
@jeppewalther that's not a bad idea - most of the actual reverse engineering was done by @joshblake @christiankerl and @sh0, so this is definitely a suggestion for those guys. 32C3 in Hamburg next winter might be a nice venue :-)

HenningJ, 2015-06-12:
But some more documentation on the API would be nice as well. Like: does the C-API in `libfreenect2/include/libfreenect2.h` actually relevant in any way? Or that the `FrameListener::onNewFrame()` callback has to delete the frame pointer once it's done with the frame. Or a better separation between the API and the internal implementation details. The actual API is just the stuff in `libfreenect2.hpp` and `frame_listener.hpp`, right?
You can figure all of this out by looking at the code, but it would be easier with some basic documentation. I'd be happy to help, but I'm not quite sure about any of it yet.

floe, 2015-06-14:
Yes, and that's a good point: `libfreenect2.hpp` and `frame_listener.hpp` should be installed by cmake to a separate include directory. If they are properly commented (which they absolutely are not at the moment), then that might be enough documentation for the first release...

xlz, 2015-10-25:
What's the standard way to generate doxygen docs?

@alberth can you provide a cmake script in cmake_modules to generate doxygen docs, maybe starting from find_package(doxygen)? Is there any style.css or script.js needed?

Then where is it to be hosted? I guess gh-pages is the easiest.

floe, 2015-11-06:
(Mostly) fixed by #450.

closed 2015-11-06 by floe

xlz, 2015-11-16:
This is not finished yet. Items left for 0.1:

* API documentation - doxygen it
* Address this complaint #439
* Odroid XU4 platform setup
* OpenCL platform setup on Intel/Nvidia dual graphics - close this #109

xlz, 2015-11-21:
PR posted in #477.