OpenKinect/libfreenect2 #288: Protonect will not shutdown

mcevoyandy, 2015-06-11:
when I hit Esc in the RGB screen or when I use Ctrl-C in the terminal the process never shutsdown. The camera windows will stop updating, I'll see the following in the terminal:

[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...

Then it just sits there. Looking in the system monitor the process is "uninterruptible" and I must restart my computer to kill it. Any thoughts on why this might be happeing on my machine?

wiedemeyer, 2015-06-12:
There are many other ways to stop a process. You could try `CTRL + \` or kill it with `kill ...`. Or kill it with a task monitor.

mcevoyandy, 2015-06-12:
Hi, Thanks for the response, but none of those methods work. From what I've
found on google it's because this process get's labeled as
'uninterruptible.' I was unable to find any solutions for how to end the
process with out powering down.

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

xlz, 2015-06-12:
What does ``strace -f -p `pidof Protonect` `` show? (Where does it hang?) Also, is there anything unusual in dmesg?

mcevoyandy, 2015-06-12:
terminal output for:
protonect - https://dl.dropboxusercontent.com/u/65115630/protonect
strace - https://dl.dropboxusercontent.com/u/65115630/strace
dmesg - https://dl.dropboxusercontent.com/u/65115630/dmseg

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

xlz, 2015-06-12:
The last line in strace log

```
[pid  3115] poll([{fd=11, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLOUT}, {fd=24, events=POLLOUT}], 4, 100) = 0 (Timeout)
```
Was there anything different after this or did it hang here?

Assuming the same got printed in a loop, apparently it was stuck in `libusb_set_interface_alt_setting()` from `UsbControl::setIrInterfaceState()`

```
[pid  3112] ioctl(24, USBDEVFS_SETINTERFACE <unfinished ...>
```

Being unable to kill points to kernel problems. But I don't know more beyond this.

mcevoyandy, 2015-06-12:
That was the last line... it just kept repeating

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

floe, 2015-06-14:
What kernel are you using? If it's older than 3.16, maybe try an update; many USB-related issues were fixed there.

mcevoyandy, 2015-06-16:
'$ uname -a' gives 3.16.0-30

Is it possible that virtualbox has anything to do with it? This is the only
difference that I can find between the machines that I've setup on
previously and this one.

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

xlz, 2015-06-16:
If you are not running Protonect inside Virtualbox then it should have nothing to do with it.

mcevoyandy, 2015-06-16:
I didn't think so, but see messages about the kernel and virtualbox while
shutting this computer down...

Andy McEvoy

Research Assistant
Correll Lab
Department of Computer Science
University of Colorado at Boulder

(832) 439 - 6150

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe