OpenKinect/libfreenect2 #291: Missing libfreenect2/CMakeLists.txt

hanyazou, 2015-06-14:
Do you have some plan to move or add CMakeLists.txt in the top directory, libfreenect2/ ?

I'm working on my OpenNI2-Freenect2Driver to integrate with the libfreenect2. In the integration, I'd like to connect my local CMakeLists.txt in OpenNI2-Freenect2Driver/ subdir to the libfreenect2's project CMakeLists.txt. But cmake does not allow to refer a CMakeLists.txt out of project top directory.

I found that only one line might be enough for now. Just "ADD_SUBDIRECTORY(examples/protonect)" in libfreenect2/CMakeLists.txt works well for my purpose.

I need your advice because I'm not a expert of cmake.

HenningJ, 2015-06-15:
> But cmake does not allow to refer a CMakeLists.txt out of project top directory.

Actually, it does. You just have to add a second parameter for the build dir of that "sub"-directory. Is just added the following to `examples/protonect/CMakeLists.txt`:
```
add_subdirectory("../../OpenNI2-Freenect2Driver" "../../../OpenNI2-Freenect2Driver/build")
```

hanyazou, 2015-06-15:
@HenningJ It works for me perfectly. Thanks a lot.

closed 2015-06-15 by hanyazou