OpenKinect/libfreenect2 #295: Is file src/test_opengl_depth_packet_processor.cpp still necessary?

laborer2008, 2015-06-16:
According to CMakeLists.txt
```
IF(ENABLE_OPENGL)
  FIND_PACKAGE(GLFW3)
  FIND_PACKAGE(OpenGL)
  IF(GLFW3_FOUND)
    INCLUDE_DIRECTORIES(${GLFW3_INCLUDE_DIRS})

    LINK_DIRECTORIES(${GLFW3_LIBRARY_DIRS})
    LIST(APPEND LIBRARIES
      ${GLFW3_LIBRARIES}
      ${OPENGL_gl_LIBRARY}
    )
  SET(LIBFREENECT2_WITH_OPENGL_SUPPORT 1)
  LIST(APPEND SOURCES
    src/flextGL.c
    src/opengl_depth_packet_processor.cpp
  )
  
  LIST(APPEND RESOURCES
    src/shader/debug.fs
    src/shader/default.vs
    src/shader/filter1.fs
    src/shader/filter2.fs
    src/shader/stage1.fs
    src/shader/stage2.fs
  )
  ENDIF()
ENDIF(ENABLE_OPENGL)
```

File test_opengl_depth_packet_processor.cpp is not added to SOURCES and not linked. Seems it's deprecated some time ago and should be removed from the master branch.

laborer2008, 2015-06-16:
Now I see that this is another program. If this file is still important seems that another project should be created with its own CMakeLists.txt

gaborpapp, 2015-06-16:
See #169. It's not necessary for most users, but if the opengl packet processor is not working for some reason on your machine it is very useful for debugging.

xlz, 2016-01-21:
@gaborpapp Same question: is this still necessary? It seems the better way to debug OpenGL is to add lots of error checking.

xlz, 2016-02-06:
I will remove this file.

closed 2016-02-06 by xlz