OpenKinect/libfreenect2 #296: skipping depth packet

KahnShi, 2015-06-16:
Hi everyone,
I do all the things according to the readme.md.
However when I run ./Protonect, 
if I use ./Protonect cpu, both rgb and depth image could work, but the output prints that
```
[DepthPacketStreamParser::onDataReceived] skipping depth packet
```

if I use ./Protonect cl, it failed, and output prints that
````
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/Documents/sources/libfreenect2/examples/protonect/lib/libfreenect2.so)
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:11 with serial 022519243547
[Freenect2Impl] found 1 devices
[OpenCLDepthPacketProcessor::init] ERROR: clGetPlatformIDs(-1001)
terminate called after throwing an instance of 'cl::Error'
  what():  clGetPlatformIDs
Aborted (core dumped)
```
if I use ./Protonect gl, it failed, and output prints that
```
./Protonect: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1: no version information available (required by /home/Documents/sources/libfreenect2/examples/protonect/lib/libfreenect2.so)
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 11 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:11 with serial 022519243547
[Freenect2Impl] found 1 devices
Segmentation fault (core dumped)
```
I have 2 questions.
1. Is everything OK if it prints "skipping depth packet" in CPU mode, although I could get both rgb and depth image?
2. How could I make cl or gl mode run? It is important to me because CPU mode is too slow.. I tried to add sudo before ./Protonect as others` advice, but it doesn`t work.
THX!

xlz, 2015-06-16:
1. It is received 30 depth packets per second. If you don't see 30 warnings per second, it is OK.
2. Your OpenCL is not installed correctly. There is not enough information for more diagnostics.

KahnShi, 2015-06-26:
@xlz  Thank you, man!
All problems are solved now.

closed 2015-06-26 by KahnShi

Abbyls, 2016-08-29:
@KahnShi Can you tell me how?

whatisor, 2017-06-20:
@xlz ,can you explain why : "It is receiving 30 depth packets per second. If you don't see 30 warnings per second, it is OK."?
If skip 15 frames, FPS becomes 15FPS, not 30FPS => is it bad,right?

I got this issue with OpenGL pipeline when we move some windows (Protonect window or other program windows)

xlz, 2017-06-20:
User asked about CPU/gl mode. The CPU/gl mode is meant for verification and its performance is bad, so getting any data at all is ok.

whatisor, 2017-06-21:
I see, thanks.
About window issue I said, do you think why OpenGL processing become so slow wen we interact with other application windows?
I have a powerful PC, I saw you log it even 120Hz, it is too big for Kinect device, isn't it? It should be maximum 30Hz because of Hardware limitation.
If by some reasons, you require it 120Hz, it will be affected by other programs which make GPU busy.