OpenKinect/libfreenect2 #297: Segmentation fault after running same code of different machine

anshulvj, 2015-06-16:
So for the past few days I've been modifying libfreenect on my (sorta)embedded machine. I have added pcl code to libfreenect2 because I wanted to visualize point cloud using cloud_viewer. The same code (with modifications to CMakeLists.txt for including PCL, Protonect.cpp, Registration.cpp and Registration.h to access point cloud) works fine on the embedded machine (well, by fine I mean that the code runs without error but I cannot see the point cloud in the window, it's all black with the rgb axes showing in one corner).

So I took the same code to a proper CPU running ubuntu 15.04 and tried to run it over there but I got the following error:

anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ sudo ./bin/Protonect gl
modprobe: FATAL: Module nvidia not found.
anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ sudo ./bin/Protonect cpu
modprobe: FATAL: Module nvidia not found.
anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ ./bin/Protonect cpu
Segmentation fault (core dumped)
anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ ./bin/Protonect cl
Segmentation fault (core dumped)
anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ ./bin/Protonect gl
Segmentation fault (core dumped)
anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ 

No matter what option I used. I also had to install opencl, pcl, opencv before I could properly compile libfreenect2. I also re-cloned libfreenect2 (fresh code) and it only ran properly with the cl argument (for any other argument or no argument it shows only color and does not show depth, ir, undistorted and registered). 

What's noteworthy is that I do not have a Nvidia graphics card so when I run as su I don't know why it's looking for Nvidia module. Before I had installed opencl and used sudo, it simply used to quit without any output. Is there anything obviously wrong I'm doing here?

More info:
I've installed pcl-1.8, opencv3.0, beignet1.0.3 (opencl) from their official website. 

clinfo for me gives the following output:

anshulvj@anshulspc:~/DimApp/libfreenect2/examples/protonect$ clinfo
Number of platforms                               1
  Platform Name                                   Intel Gen OCL Driver
  Platform Vendor                                 Intel
  Platform Version                                OpenCL 1.2 beignet 1.0.3
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd
  Platform Extensions function suffix             Intel

  Platform Name                                   Intel Gen OCL Driver
Number of devices                                 1
  Device Name                                     Intel(R) HD Graphics Haswell GT2 Desktop
  Device Vendor                                   Intel
  Device Vendor ID                                0x412
  Device Version                                  OpenCL 1.2 beignet 1.0.3
  Driver Version                                  1.0.3
  Device OpenCL C Version                         OpenCL C 1.2 beignet 1.0.3
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Max compute units                               20
  Max clock frequency                             1000MHz
  Device Partition                                (core)
    Max number of sub-devices                     1
    Supported partition types                     None, None, None
  Max work item dimensions                        3
  Max work item sizes                             512x512x512
  Max work group size                             512
  Preferred work group size multiple              16
  Preferred / native vector sizes                 
    char                                                16 / 8       
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 0 / 8        (n/a)
    float                                                4 / 4       
    double                                               0 / 2        (n/a)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
[... 35 lines cut ...]
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Device Extensions                               cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Intel Gen OCL Driver
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [Intel]
  clCreateContext(NULL, ...) [default]            Success [Intel]
  clCreateContext(NULL, ...) [other]              <error: no devices in non-default plaforms>
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Intel Gen OCL Driver
    Device Name                                   Intel(R) HD Graphics Haswell GT2 Desktop
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Intel Gen OCL Driver
    Device Name                                   Intel(R) HD Graphics Haswell GT2 Desktop

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.3
  ICD loader Profile                              OpenCL 1.2

xlz, 2015-06-17:
Can you produce the stack trace?

```
ulimit -c unlimited
./bin/Protonect xxx
Segmentation fault (core dumped)
gdb ./bin/Protonect core.xxxx
(gdb) bt
```

anshulvj, 2015-06-24:
I changed from opencv 3 to opencv 2.4.9 and it worked.

closed 2015-06-24 by anshulvj