OpenKinect/libfreenect2 #303: Protonect keep showing pink RGB image

xionglingfeng, 2015-06-21:
I compiled latest libfreenect2 (2ea7c34199) in Mac OS X Yosemite 10.10.3. Kinect is connected to a USB 3.0 port. But Protonect keep showing a pink image in RGB.
[image]
I tried it in a Macbook Pro and Macbook Air and saw the same problem.

ThomasLengeling, 2015-06-21:
Hey I am also having the asme issue on my Mac 10.10.3.
The depth looks great, the IR and RGB Frames have the pink image

Thomas

floe, 2015-06-22:
I think this is a duplicate of an old bug which happens when you build `libturbojpeg` yourself. Please try installing from ports or brew as indicated by the Readme.

PyrApple, 2015-06-22:
Same issue, installing libturbojpeg with brew as indicated in the Readme ( for current commit, i.e. 2ea7c3419932362376fb2dbb20b690f6730624f1)

xionglingfeng, 2015-06-23:
I have tried to link libfreenect2 with homebrew's turbo-jpeg and the version in 'depends' folder. Neither of them works.

However, I remember in February, it works without any issues.

floe, 2015-06-23:
I don't have a Mac, so I can't reproduce this - please see the original bug #184 and check if any of the fixes proposed there work for you...

xionglingfeng, 2015-06-23:
It seems #184 uses external version of jpeg-turbo (from homebrew) replace the one in 'depends'. I already tried that but no help.

PyrApple, 2015-07-02:
A build from sources of jpeg-turbo based on nasm 2.11.06 solved the issue for me (OSX 10.9).

It's obviously a temporary fix, but awaiting fix on jpeg-turbo "bottle" (homebrewed pre-built on nasm 2.11.08) homebrewers can use (uninstall nasm and jpeg-turbo beforehand):

cd /usr/local/Library/Formula/
git checkout 8dd3478131b232553f4c0cab197f8443a0a559a6 jpeg-turbo.rb
git checkout db83e50df56d780176ad452371febed20cf25cbb nasm.rb
brew install jpeg-turbo --build-from-source
git reset HEAD jpeg-turbo.rb
git reset HEAD nasm.rb
git checkout nasm.rb
git checkout jpeg-turbo.rb

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe