OpenKinect/libfreenect2 #304: Multiple Kinects does not work

xionglingfeng, 2015-06-23:
I connected two Kinect2 on my computer and write a program to capture video from both Kinects.
This program creates two threads, each threads execute `Freenect2Device *dev = params->driver->openDevice(params->i, new OpenCLPacketPipeline());` to open a Kinect with device index.
When the program runs, it reports lots of errors said `
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_OTHER`. Also, only one Kinect return data, the other thread just stucked at `listener.waitForNewFrame(frames);`.
I tested this with latest version (2ea7c3419932362376fb2dbb20b690f6730624f1) on both Ubuntu 15.04 and Mac OS X 10.10.3.

antigp, 2015-08-03:
Have the same issues. Does external controller for second kinect fix this problem?

siddharthachandra, 2015-08-03:
I confirm this issue on my Ubuntu 14.04 (with built-in Texas Instruments USB 3.0 HUBs on my HP Z420).

Edwinzero, 2015-08-14:
External USB controller can solve this.

xlz, 2015-10-05:
`0xe00002ec` means `kIOReturnNoBandwidth`.

closed 2015-10-09 by xionglingfeng

rain3k, 2017-06-14:
I have same issue.....
I try to connect external usb to Mac Pro
but it didn't work........
how to solve this issue.