OpenKinect/libfreenect2 #305: Build failed when running sh install_ubuntu.sh

SamTew, 2015-06-23:
Here is my output. Can anyone help me to solve?

sam@sam-X450LCP:~/libfreenect2/depends$ sh install_ubuntu.sh
--2015-06-23 23:39:34--  http://archive.ubuntu.com/ubuntu/pool/universe/g/glfw3/libglfw3_3.0.4-1_amd64.deb
Resolving archive.ubuntu.com (archive.ubuntu.com)... 91.189.91.15, 91.189.91.13, 91.189.92.201, ...
Connecting to archive.ubuntu.com (archive.ubuntu.com)|91.189.91.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33000 (32K) [application/x-debian-package]
Server file no newer than local file ‘libglfw3_3.0.4-1_amd64.deb’ -- not retrieving.

--2015-06-23 23:39:35--  http://archive.ubuntu.com/ubuntu/pool/universe/g/glfw3/libglfw3-dev_3.0.4-1_amd64.deb
Resolving archive.ubuntu.com (archive.ubuntu.com)... 91.189.91.13, 91.189.92.201, 91.189.91.24, ...
Connecting to archive.ubuntu.com (archive.ubuntu.com)|91.189.91.13|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 19168 (19K) [application/x-debian-package]
Server file no newer than local file ‘libglfw3-dev_3.0.4-1_amd64.deb’ -- not retrieving.

Cloning into '/home/sam/libfreenect2/depends/libusb_src'...
remote: Counting objects: 10776, done.
remote: Total 10776 (delta 0), reused 0 (delta 0), pack-reused 10776
Receiving objects: 100% (10776/10776), 2.61 MiB | 873.00 KiB/s, done.
Resolving deltas: 100% (7935/7935), done.
Checking connectivity... done.
Note: checking out '51b10191033ca3a3819dcf46e1da2465b99497c2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 51b1019... linux_usbfs: Update MAX_ISO_BUFFER_LENGTH
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
configure.ac:4: installing './compile'
configure.ac:4: installing './config.guess'
configure.ac:4: installing './config.sub'
configure.ac:2: installing './install-sh'
configure.ac:2: installing './missing'
Makefile.am: installing './depcomp'
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:39: installing './config.guess'
configure.ac:39: installing './config.sub'
configure.ac:29: installing './install-sh'
configure.ac:29: installing './missing'
examples/Makefile.am: installing './depcomp'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
[... 169 lines cut ...]
 /bin/mkdir -p '/home/sam/libfreenect2/depends/libusb/include/libusb-1.0'
 /usr/bin/install -c -m 644 libusb.h '/home/sam/libfreenect2/depends/libusb/include/libusb-1.0'
make[3]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src/libusb'
make[2]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src/libusb'
make[1]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src/libusb'
Making install in doc
make[1]: Entering directory `/home/sam/libfreenect2/depends/libusb_src/doc'
make[2]: Entering directory `/home/sam/libfreenect2/depends/libusb_src/doc'
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src/doc'
make[1]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src/doc'
make[1]: Entering directory `/home/sam/libfreenect2/depends/libusb_src'
make[2]: Entering directory `/home/sam/libfreenect2/depends/libusb_src'
make[2]: Nothing to be done for `install-exec-am'.
 /bin/mkdir -p '/home/sam/libfreenect2/depends/libusb/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/home/sam/libfreenect2/depends/libusb/lib/pkgconfig'
make[2]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src'
make[1]: Leaving directory `/home/sam/libfreenect2/depends/libusb_src'

Execute the following commands to install the remaining dependencies (if you have not already done so):

sudo dpkg -i libglfw3*_3.0.4-1_*.deb
sudo apt-get install build-essential libturbojpeg libjpeg-turbo8-dev libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev automake
sudo apt-get install libturbojpeg0-dev # (Debian)

floe, 2015-06-23:
The script did exactly what it's supposed to - I don't see any error message.

floe, 2015-06-25:
Closing - not a bug.

closed 2015-06-25 by floe