OpenKinect/libfreenect2 #306: build requires at least cmake version 2.8.8 or even higher

christiankerl, 2015-06-23:
cmake 2.8.7 doesn't work because of `ADD_LIBRARY(freenect2 OBJECT ${SOURCES})` in CMakeLists.txt (see http://www.cmake.org/Wiki/CMake/Tutorials/Object_Library) 

cmake 2.8.8 didn't work because of a different error:
```
CMake Error at /usr/local/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97 (MESSAGE):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES LibUSB_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/local/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:288 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:42 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:42 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
```
latest cmake version 3.2.3 works

this should be documented somewhere

floe, 2015-06-23:
Hmm, the required minimum is probably somewhere in between - I'm currently on CMake 2.8.12.2 (default version in Ubuntu 14.04), and this works, too.

HenningJ, 2015-06-26:
It should be enough to raise the [CMake minimum required version](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/CMakeLists.txt#L1) to 2.8.12. Then everybody below that gets an error when running cmake

closed 2015-06-26 by floe