OpenKinect/libfreenect2 #307: Pausing kinect v2 stream (not a bug)

anshulvj, 2015-06-24:
This might be an uncommon question: Is there any way to pause the kinect streams? I looked at the code and we do have dev->start(), stop(), and close(). I could possibly use these as well, but is there a way to create a pause() where it doesn't shut down the device, just pauses it so that it can be restarted quickly.
The reason I ask this is that my small embedded machine conks out after a minute or so (it wedges to be more clear) and restarts the session, because obviously it cannot handle so much data. I don't want a continuous stream anyway; only when requested.

floe, 2015-06-24:
Hmm, off the top of my head, you could try to add a `pause()` function to `Freenect2DeviceImpl` which just calls `command_tx_.execute(SetStreamDisabledCommand(nextCommandSeq()), result);` and see if that is sufficient.

anshulvj, 2015-06-24:
Thanks I'll try that

anshulvj, 2015-06-24:
Just to be clear, you mean something similar to the stop() function, with just one statement :

void Freenect2DeviceImpl::pause()
{
 command_tx_.execute(SetStreamDisabledCommand(nextCommandSeq()), result);
}

anshulvj, 2015-06-24:
Ok, that seemed to work. Now my application runs without crashing the machine (although it gets hot). I will comment some more things so it's not doing any unnecessary work.

bellekci, 2015-06-27:
But this setStreamDisabledCommand does not stop IR emitters. 
It would be nice if we could stop and restart IR emitters in a fast way.

floe, 2015-06-27:
Currently, we don't know if there is such a command. Can the Windows SDK turn the emitter on and off? If it can't, then it's doubtful that we can find out how to do this.

christiankerl, 2015-06-27:
I think the Unknown0x0ACommand turns the IR LEDs off (see the stop() method) furthermore, I think SetStreamEnabledCommand / SetStreamDisabledCommand controls only RGB streaming. could you verify this?

bellekci, 2015-06-28:
No, Windows SDK  does not have such a feature either.

pammirato, 2015-08-07:
how did you get this to work?  I tried:

void Freenect2DeviceImpl::pause()
{
  CommandTransaction::Result result;
  command_tx_.execute(SetStreamDisabledCommand(nextCommandSeq()), result);
}//pause

void Freenect2DeviceImpl::unpause()
{
  CommandTransaction::Result result;
  command_tx_.execute(SetStreamEnabledCommand(nextCommandSeq()), result);
}//unpause

but when i call pause() I get this:

[DepthPacketStreamParser::handleNewData] working buffer full, resetting it!
[DepthPacketStreamParser::handleNewData] remaining data should be 8 bytes, but is 4583

I am running libfreenect2 with Kinect2Bridge, through ROS on ubuntu, but I think that error message is from libfreenect2

xlz, 2016-01-02:
I believe what you need is in fact just stop() and start(). Protonect.cpp has an examples of how to use that. Consider this implemented.

closed 2016-01-04 by floe