OpenKinect/libfreenect2 #310: Cant install libfreenect2 on mac

spurra, 2015-06-28:
Hi,
I've followed the installation instruction here https://github.com/OpenKinect/libfreenect2#mac-osx. The step to build the protonect executable with cmake fails with following error:

-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
CMake Error at /usr/local/Cellar/cmake/3.2.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES LibUSB_INCLUDE_DIRS)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.2.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:374 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:42 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:42 (FIND_PACKAGE)

Are there some other things I should have installed beforehand which the guide does not mention?

floe, 2015-06-29:
Looks like you are missing most of the generic build tools (`autoconf`,`pkg-config` etc.) I'm not a Mac expert, so I don't know what the recommended way of installing them would be, but I assume you need something like Xcode + commandline tools installed first.

spurra, 2015-06-29:
I'll see to it that I can install these build tools. However, again, in the shell script install_libusb.sh it tries to invoke bootstrap.sh which does not exist in the subfolder.

floe, 2015-06-29:
Yes, because bootstrap.sh is part of the libusb code which is downloaded in https://github.com/OpenKinect/libfreenect2/blob/master/depends/install_libusb.sh#L13

xionglingfeng, 2015-06-30:
I think you'd better install [homebrew](http://brew.sh/) to default location or add `homebrew/bin` to `$PATH`.

spurra, 2015-06-30:
floe, I have libusb installed via brew. However, I installed the other dependencies and it seemed to have compiled fine. It runs well when using the parameter CPU, however, it crashes using my NVidia GPU. Are there any known issues with that?
xionglingfeng, I've installed homebrew via the ruby command line, so I imagine its installed in the default location.

xionglingfeng, 2015-06-30:
Hi spurra,
There is a know bug (#205) affecting OpenGL processor. Try to use OpenCL (./Protonect cl) instead.

xcottcraver, 2015-07-17:
Hi all, 

I just spent an afternoon trying to get Protonect installed and working from the instructions, encountering many of the issues listed here and in other threads. 

I ultimately got it working by following the instructions in the README.md file, but between steps 2 and 3 I did a "brew install" for the packages:  cmake (ha ha I didn't have cmake!) pkg-config, automake and libtool.  These are machines that already had Xcode and command-line tools on them.  

I also replaced the "sh ./depends/install_mac.sh" in step 4 with "sudo sh ./depends/install_mac.sh", and in step 5 I replaced the "make && make install" with a make followed by a sudo make install.

A few other notes:

1.  I did *not* brew install libusb, because install_mac.sh does it, and I didn't want to mess with that.  I have no idea if installing it separately would be an issue.
2.  simply brew installing those missing packages doesn't seem to help if you already tried to run the install_mac script.  I got rid of the libfreenect2 I had, got those packages, and repeated everything with a fresh git.
3.  This did not work on 10.10.3.  I could get Protonect compiled, but it would crash when run.  That's an entirely separate issue, but the above instructions produced a Protonect that worked on several machines running 10.9.5

xlz, 2015-07-17:
> brew install libusb

OK, here is the thing. I think we have to explicitly ask the users not to do this in README.md, before libusb is built statically by #283. 

> ha ha I didn't have cmake!

I assumed cmake, pkg-config, automake, autoconf, and libtool must be installed by Xcode or some kind of toolchains. Perhaps for the sake of completeness, also add instructions for these packages in README.

> sudo sh ./depends/install_mac.sh

You don't have to do this. `./depends/install_mac.sh` has already included `sudo` for the command that needs it.

hanyazou, 2015-07-17:
I tried to build protonect on clean installed Mac OS X 10.10 on VMware and I found that brew install wget is also required.

floe, 2015-07-18:
Can someone with a Mac perhaps send a PR updating/correcting the README?

braitsch, 2015-08-11:
I'm having some trouble here too, failing at step 6: 

    cmake ..

```
-- checking for module 'libusb-1.0'
--   package 'libusb-1.0' not found
CMake Error at /usr/local/Cellar/cmake/3.3.0/share/cmake/Modules/FindPkgConfig.cmake:340 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.3.0/share/cmake/Modules/FindPkgConfig.cmake:502 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:15 (PKG_CHECK_MODULES)
  CMakeLists.txt:42 (FIND_PACKAGE)
```

When I run `sh ./depends/install_mac.sh` it's building libusb in `./libfreenect2/depends/libusb` so not sure what's wrong here. Any ideas?

xlz, 2015-08-11:
@braitsch Apply this PR #357.

braitsch, 2015-08-11:
@xlz That worked, thanks!

floe, 2015-08-12:
Closed by #357.

closed 2015-08-12 by floe

v2k, 2016-04-14:
@xlz still having this issue in the latest version, following the instructions in the Readme for OSX.  Am I missing something?

xlz, 2016-04-14:
The above is an invalid report.

ericwenger7, 2017-02-02:
libfreenect2 does no longer contain an install for mac. 
sh ./depends/install_mac.sh
sh: ./depends/install_mac.sh: No such file or directory

Any workaround ? 

still no luck at using a kinect one on mac osx 10.11

xlz, 2017-02-02:
> ./depends/install_mac.sh

Read the instructions.

ericwenger7, 2017-02-02:
You mean the README ? I did that first but trying to cmake I get the following errors : /Applications/CMake.app/Contents/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:102 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:93 (FIND_PACKAGE)

floe, 2017-02-03:
Update your repo to the latest version, then read https://github.com/OpenKinect/libfreenect2/blob/master/README.md#mac-osx

ericwenger7, 2017-02-03:
I re-run the whole sequence and now it works fine !  Thanks

ericwenger7, 2017-02-04:
However all of the openni2 apps fail to open a window and display an image under mac osx 10.11.6.
the device is found and works fine in the terminal console. Menus of NiViewer are not working either and one need to force quit. 
Protonect works fine.
Any idea why?