OpenKinect/libfreenect2 #311: OpenNI2 driver resolution error

xionglingfeng, 2015-06-29:
I compiled libfreenect2 with OpenNI2 driver from [this fork](https://github.com/hanyazou/libfreenect2) on Mac OS X. `src/DeviceDriver.cpp` has been modified to use OpenCL driver instead of OpenGL to workaround a recent bug(#205) in `libfreenect2`.
When I execute `./SimpleViewer`, I got:

    [Freenect2DeviceImpl] enabling usb transfer submission...
    [Freenect2DeviceImpl] submitting usb transfers...
    [Freenect2DeviceImpl] started
    [Freenect2DeviceImpl] starting...
    Error - expect color and depth to be in same resolution: D: 640x480, C: 1920x1080
    [Freenect2DeviceImpl] stopping...
    [Freenect2DeviceImpl] disabling usb transfer submission...
    [Freenect2DeviceImpl] canceling usb transfers...
    [DepthPacketStreamParser::onDataReceived] not all subsequences received 0
    [Freenect2DeviceImpl] stopped
    [Freenect2DeviceImpl] stopping...
    [Freenect2DeviceImpl] already stopped, doing nothing
    OpenNI2-Freenect2Driver: Closing device freenect2://0

`pcl_openni2_viewer` does not work either with:

    [Freenect2DeviceImpl] enabling usb transfer submission...
    [Freenect2DeviceImpl] submitting usb transfers...
    [Freenect2DeviceImpl] started
    [Freenect2DeviceImpl] starting...
    [Freenect2DeviceImpl] starting...
    Device ID not set, using default device: Kinect_Microsoft
    [Freenect2DeviceImpl] stopping...
    [Freenect2DeviceImpl] disabling usb transfer submission...
    [Freenect2DeviceImpl] canceling usb transfers...
    [DepthPacketStreamParser::onDataReceived] not all subsequences received 0
    [Freenect2DeviceImpl] stopped
    [Freenect2DeviceImpl] stopping...
    [Freenect2DeviceImpl] already stopped, doing nothing
    [Freenect2DeviceImpl] stopping...
    [Freenect2DeviceImpl] already stopped, doing nothing
    OpenNI2-Freenect2Driver: Closing device freenect2://0

I am confused whats happening here? Any suggestion?

hanyazou, 2015-06-29:
Hi. Thank you for trying on the driver. It seems that SimpleViewer does not support the driver's default resolution and there is no way to change the resolution. Could you try NiViewer instead? I've tried NiViewer and It worked with the default resolution of the driver.

xionglingfeng, 2015-06-30:
Hi hanyazou,
Thank you for your suggestion. It seems `NiViewer` works. However, PCL still not working here. As my thread said, it reports `Device ID not set, using default device: Kinect_Microsoft`, then stopped.

hanyazou, 2015-06-30:
@xionglingfeng, because I was afraid that something has been broken, I tried updated sources today. I found that every thing looks ok then.

1. I've updated my local clones from repositories, today Web Jul 1, 2015

  https://github.com/occipital/OpenNI2
  https://github.com/PointCloudLibrary/pcl
  https://github.com/hanyazou/libfreenect2

2. checked out master branch.

3. patched locally
diff --git a/ThirdParty/PSCommon/BuildSystem/Platform.x86 b/ThirdParty/PSCommon/BuildSystem/Platform.x86
index 04dd480..b5ba042 100644
--- a/ThirdParty/PSCommon/BuildSystem/Platform.x86
+++ b/ThirdParty/PSCommon/BuildSystem/Platform.x86
@@ -7,8 +7,8 @@ endif

 ifeq ("$(OSTYPE)","Darwin")
        # Making the binary a universal one (x86 + x64)
-       CFLAGS += -arch i386 -arch x86_64
-       LDFLAGS += -arch i386 -arch x86_64
+       #CFLAGS += -arch i386 -arch x86_64
+       #LDFLAGS += -arch i386 -arch x86_64
 endif

 ifeq ($(SSE_GENERATION), 2)

diff --git a/visualization/tools/openni2_viewer.cpp b/visualization/tools/openni2_viewer.cpp
index 595e5a9..d0af77e 100644
--- a/visualization/tools/openni2_viewer.cpp
+++ b/visualization/tools/openni2_viewer.cpp
@@ -296,6 +296,7 @@ main (int argc, char** argv)
   pcl::io::OpenNI2Grabber::Mode image_mode = pcl::io::OpenNI2Grabber::OpenNI_Default_Mode;
   bool xyz = false;

+#if 0
   if (argc >= 2)
   {
     device_id = argv[1];
@@ -341,6 +342,7 @@ main (int argc, char** argv)
       cout << "Device ID not set, using default device: " << device->getStringID () << endl;
     }
   }
+#endif // 0

   unsigned mode;
   if (pcl::console::parse (argc, argv, "-depthmode", mode) != -1)

4. just rebuild them
  cd ~/github/pcl && make
  cd ~/github/OpenNI2 && make
  cd ~/github/libfreenect2/examples/protonect
  cmake -DENEBLE_OPENGL=OFF .
  make
  cd ~/github/libfreenect2/drivers/OpenNI2-Freenect2Driver
  cmake .
  make

I could see point cloud as I expected. Both pcl_openni2_viewer.app and  pcl_openni2_viewer.app/Contents/MacOS/pcl_openni2_viewer work well.

My environment is 

bash-3.2$ sw_vers
ProductName:    Mac OS X
ProductVersion: 10.9.5
BuildVersion:   13F1077
bash-3.2$ xcodebuild -version
Xcode 6.2
Build version 6C131e
bash-3.2$ c++ --version
Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin13.4.0
Thread model: posix

Are there any difference ?

xionglingfeng, 2015-07-02:
Hi @hanyazou . I built `pcl` and `OpenNI2` with `homebrew`. Here's the build info:

    brew info openni2
    homebrew/science/openni2: stable 2.2.0.33, HEAD
    http://structure.io/openni
    /Volumes/Application/homebrew/Cellar/openni2/2.2.0.33 (151 files, 9.1M) *
      Built from source
    From: https://github.com/homebrew/homebrew-science/blob/master/openni2.rb

    homebrew/science/pcl: stable 1.7.2, HEAD
    http://www.pointclouds.org/
    /Volumes/Application/homebrew/Cellar/pcl/1.7.2 (1084 files, 123M) *
      Built from source with: --with-openni2 --with-examples
    From: https://github.com/homebrew/homebrew-science/blob/master/pcl.rb

`libfreenect2` is on master branch (ff81f5ca76778c44bb8bafcdcbb307f2182ef582)

hanyazou, 2015-07-02:
> I think those patches are irrelevant to this issue?

I agree. I just mean that I didn't modify PCL and OpenNI2.

PCL 1.7.2 was released on Sep 11, 2014. It looks little bit old.

closed 2015-11-09 by floe