OpenKinect/libfreenect2 #313: opencl headers

giacomodabisias, 2015-07-06:
Hi,
when I try to compile the project I get:
opencl_depth_packet_processor.cpp:250:12: error: ‘CL_DEVICE_TYPE_CUSTOM’ was not declared in this scope

I used already libfreenect2 and I could fix the problem without a big effort, by just removing that, but I would prefer to fix it. I install the opencl-headers package, but nothing.

Any ideas?

floe, 2015-07-06:
On Ubuntu 14.04, my version of opencl-headers is 1.2-2013.10.23-1, and this does contain `#define CL_DEVICE_TYPE_CUSTOM (1 << 4)` in cl.h. Can you see if you maybe have an older package revision?

xlz, 2015-07-08:
With Nvidia? Nvidia's OpenCL headers are old. A very easy fix is to add `#ifdef CL_DEVICE_TYPE_CUSTOM` around CL_DEVICE_TYPE_CUSTOM.

wiedemeyer, 2015-07-09:
I have an Ubuntu 14.04 with a Nvidia GPU. I have the following packages installed and don't have any issues, `nvidia-352`, `opencl-headers`, `ocl-icd-libopencl1`, `nvidia-opencl-icd-352` (the last two are installed automatically)

giacomodabisias, 2015-07-09:
Got the same packages, but no one of them installed libopencl.so, strange

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe