OpenKinect/libfreenect2 #316: color, depth registration issue

flyfj, 2015-07-09:
I used the example program from protonect. It has no problem fetching rgb and depth data, but when i call the registration function, the returned color is not well aligned.
the code i used:
```
libfreenect2::Registration* registration = new libfreenect2::Registration(dev->getIrCameraParams(), dev->getColorCameraParams());
...
registration->apply(rgb, depth, registered);
...
```
example images are shown below:
[image]
[image]

Official SDK doesn't have this problem if I remember it correctly. Struggling with this issue now.

floe, 2015-07-10:
This is a bit weird, never seen such a large mismatch so far. With libfreenect 2 on my Kinect devices, the images line up very well. Can you check if you have the latest master branch?

wiedemeyer, 2015-07-10:
It cant be the newest master, the new api for the registration looks like this now:
`registration->apply(rgb,depth,&undistorted,&registered);`

flyfj, 2015-07-10:
@floe @wiedemeyer thanks for the advice. didn't notice the new branch change. will update and test.

flyfj, 2015-07-10:
@floe So I rebuilt from the latest master branch, but it still gives me the same results. From the command line window, it shows messages like: skipping depth package. subpackage too large.
It looks like the registration is running late behind the fetching or display.

floe, 2015-07-11:
Was the screenshot taken with a moving camera? You could try a different depth packet processor, see here: https://github.com/OpenKinect/libfreenect2/blob/master/README.md#im-seeing-the-color-camera-stream-but-no-depthir-black-windows

floe, 2015-07-22:
Any news? If there's still need for discussion, please re-open.

closed 2015-07-22 by floe