OpenKinect/libfreenect2 #319: Warped pointcloud

robinJKU, 2015-07-14:
Hello,

I have a strange issue regarding my kinect2, when facing down about 45° at a table I get a warped pointcloud, it looks pretty much the same with and without calibration.
I calibrated the kinect using a checker board with nearly 200 images.

Setup:
[image]
Pointcloud:
[image]

Is this a ToF issue or is it possible to fix the problem?

JoshBlake, 2015-07-14:
This looks like the interior corner reflection artifact common to ToF
sensors. The laser pulse illuminates the scene, but the diffuse surfaces
sometimes reflect the pulse back across multiple bounces (off the
checkerboard, then off the table, then to the sensor, or table then
checkerboard). This input is added to the true single-reflection data and
is smoothed within the sensor stack to give you that warped surface.

There isn't much that you can do unless you want to post-process the data
manually, or change the scene geometry to avoid interior corners, or change
the scene materials to have at least one un-desired interior plane use a
darker fabric to absorb more of the IR pulse reflection.

---
Joshua Blake
Co-founder & VP of Engineering
Orbbec http://www.orbbec3d.com

(cell) 703-946-7176
Twitter <http://twitter.com/joshblake> | Blog <http://nui.joshland.org>

robinJKU, 2015-07-14:
Thanks for your reply, I already tried moving the table away from the wall and get the same result.

floe, 2015-07-14:
If it's not a ToF hardware issue, then it might be related to #144 - do you have an Windows 8 machine at hand and could try to extract the xTable/zTable offset maps from the GPU?

cnanda-zbr, 2016-10-12:
This is the effect of multipath.