OpenKinect/libfreenect2 #320: How many Kinect V2 a USB 3.0 controler support?

yuanmingze, 2015-07-14:
My PC's chipset is Intel® Z97 Chipset.There are one USB 3.0 control.
1.When connecting 2 Kinect V2s to pc it works well. But when connecting the 3th Kinect it doesn't work. Can I add a PCI-E USB 3.0 controler to fix it ?
2.Readme say “libfreenect2 work for up to 5 devices”,how to connect 5 Kinect V2s with PC ?

floe, 2015-07-14:
1. Yes, probably, although it's unpredictable if a specific hardware combination will actually work.
2. This was done by @steppobeck using 4 PCIe USB controller cards plus the one on the mainboard. Maybe he can provide some detail on the hardware he used.

floe, 2015-07-14:
Some related discussion also in #211.

steppobeck, 2015-07-14:
Yes, I can confirm that each Kinect needs its own USB-controller. Current main boards typically have one or two USB 3.0 controllers, so one needs to plug in USB 3.0 expansion card(s). In my current setup I use 4 x Transcend TS-PDU3 USB3.0 Expansion Cards and one on board controller. Running Ubuntu 14.04 LTS.

Kind regards,
Stephan Beck

--

Stephan Beck
Virtual Reality Systems Group
Faculty of Media
Bauhaus-Universitaet Weimar
Bauhausstra�e 11
99423 Weimar
Germany
phone: +49 3643 - 58 3734
fax  : +49 3643 - 58 3709
email: stephan.beck@uni-weimar.de
email: stephan@pixelstars.de
web  : http://www.uni-weimar.de/de/medien/professuren/vr/

floe, 2015-07-14:
@steppobeck thanks, very helpful - will include that in the FAQ.

yuanmingze, 2015-07-14:
@steppobeck thanks a lot.
I want to use one PC taking at least 4 Kinect V2s . My PC configuration is 
CPU : Intel Core i7 4790K
Display adapter : NVIDIA GeForce GTX 980
Memory : 32G
1.Can the PC  take 4 Kinect v2s to capture Deep information and video in 30fps?
2.Could tell me whether other kind USB3.0 Expansion Cards work well,except TS-PDU3?

floe, 2015-07-14:
IMHO the PC should be powerful enough. For 2., see the FAQ: https://github.com/OpenKinect/libfreenect2#im-getting-lots-of-usb-transfer-errors-andor-only-blank-windows

steppobeck, 2015-07-14:
Hi,

> @steppobeck thanks a lot.
> I want to use one PC taking at least 4 Kinect V2s . My PC configuration is 
> CPU : Intel Core i7 4790K
> Display adapter : NVIDIA GeForce GTX 980
> Memory : 32G
> 1.Can the PC take 4 Kinect v2s to capture Deep information and video in 30fps?
I think yes! However, it depends on what other processing has to be performed. Depth-only @ 30 Hz should be fine with your PC.
> 2.Could tell me whether other kind USB3.0 Expansion Cards work well,except TS-PDU3?
I have only tested the TS-PDU3. These cards should work too:
Cards confirmed working here on the forums:
	� SIIG Inc. DP 2-Port USB 3.0 PCIe (JU-P20612-S1), see http://www.siig.com/it-products/usb/adapters/pcie/dp-2-port-usb-3-0-pcie.html . Available in US
	� Transcend TS-PDU3 USB3.0 Expansion Card. Available in Europe. http://www.amazon.com/Transcend-Schnittstellenkarte-Adapter-PCIe-Ports/dp/B003MVJG8Q
	�  2 Port USB 3.0 HUB to PCI-E PCI Express Card Converter Adapter 5.0Gbps Renesas D720200.http://www.amazon.co.uk/gp/product/B008R535MU/ref=s9_simh_gw_p147_d0_i1?pf_rd_m=A3P5ROKL5A1OLE&pf_rd_s=center-2&pf_rd_r=1Y0RGAR9ENQSY1S8GHZY&pf_rd_t=101&pf_rd_p=430153987&pf_rd_i=468294 , available in UK.

Kind regards,
Stephan

> 
> �

--

Stephan Beck
Virtual Reality Systems Group
Faculty of Media
Bauhaus-Universitaet Weimar
Bauhausstra�e 11
99423 Weimar
Germany
phone: +49 3643 - 58 3734
fax  : +49 3643 - 58 3709
email: stephan.beck@uni-weimar.de
email: stephan@pixelstars.de
web  : http://www.uni-weimar.de/de/medien/professuren/vr/

yuanmingze, 2015-07-15:
It's very helpful. Thanks a lot.

closed 2015-07-15 by yuanmingze

jasoncatchpole, 2015-07-15:
Thanks for the info guys.

One question: Do you think the limit of 5 Kinects is based on running up against PCIe bandwidth limitations, or the ability to process that amount of depth stream data? 

With current chipsets there is only 4xPCIe 2.0 (about 20Gbps) bandwidth from southbridge to CPU so maybe that is the limiting factor that you are running into with 5 Kinects?

For instance if one were to use one of the new Skylake processors with the new Z170 chipset which moves from PCIe Gen2.0 slots to Gen3.0 slots off the southbridge and moves from a DMI Gen2.0 to a Gen3.0 connection from southbridge to CPU (so basically double the bandwidth), would this help get to larger numbers of cameras?

floe, 2015-07-16:
I don't think that 5 is a hard limit, I've also read reports of 6 devices on one PC (although I can't remember the exact details right now). However, due to USB3 still being pretty flaky IMHO, be prepared for a bit of trial and error...

jasoncatchpole, 2015-07-16:
Cool, thanks for the details.

xlz, 2015-10-24:
4xPCIe 2.0 is 16Gbps. Each Kinect requires about 3Gbps bandwidth reservation. So 5 kinects would leave very little bandwidth left on your PCIe bus.

Kinect 2's data rate is smaller than its bandwidth reservation, about 1 Gbps (120MB/s). There would be PCIe round trips to the GPU too... Processing 5*120MB/s along seems doable.

alexjheck, 2019-02-21:
I have had success running multiple kinect v2s using StarTech's Part# PEXUSB3S4V, a 4 port USB 3.0 PCIe card with an individual controller for each port.

1gen100, 2019-10-02:
memory     	32GB 
processor  	Intel(R) Core(TM) i7-4910MQ CPU @ 2.90GHz
/0/100                 	         bridge     	Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller
/0/100/1               	         bridge     	Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller
display    	        4th Gen Core Processor Integrated Graphics Controller
/0/100/3               	         multimedia 	Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller

would this work for a laptop setup:
https://www.amazon.com/Transcend-Express-Adapter-Laptop-TS-PNU3/dp/B003QJX1K8