OpenKinect/libfreenect2 #321: [Ubuntu 14.04] Protonect no window shown after running

renxinhe, 2015-07-15:
I tried running Protonect on my Ubuntu 14.04 today with `sudo ./bin/Protonect`, and this is the output:
> [Freenect2Impl] enumerating devices...
> [Freenect2Impl] 14 usb devices connected
> [Freenect2Impl] found valid Kinect v2 @4:9 with serial 501314743042
> [Freenect2Impl] found 1 devices

The command outputs seem just stopped there without displaying any additional lines. It automatically returned back to bash. I think I'm supposed to see some new Protonect windows opening and showing images. However, none of those showed up and seems like the program just halted.

floe, 2015-07-15:
Sounds like an USB issue, what's your kernel version & USB controller? Please also see the FAQ.

qiuxt, 2015-07-16:
I also meet with this problem. But when I try sudo ./Protonect cpu , it works well. So I think it may not be a USB issue.

qiuxt, 2015-07-16:
It fixed by rebooting...

renxinhe, 2015-07-16:
This is my kernel version: `3.16.0-43-generic`.

This is output from running `sudo ./bin/Protonect`:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 14 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened
[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 20 00 17 ba 00 08 00 00 10 00 00 00    2e213355202e2e2e2e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 02 0b 08 47 4b 52 35 32 33 2e 31    31332e2e2e2e2e2e474b523532332e31
0x0050:  58 00 00 00 00 00 00 00 00 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

>[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[... 14 lines cut ...]
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[TransferPool::submit] failed to submit transfer: LIBUSB_ERROR_IO
[Freenect2DeviceImpl] started
device serial: 501314743042
device firmware: 4.3.3912.0.7
[TurboJpegRgbPacketProcessor] avg. time: 15.289ms -> ~65.4065Hz
[TurboJpegRgbPacketProcessor] avg. time: 14.1246ms -> ~70.7985Hz
[TurboJpegRgbPacketProcessor] avg. time: 13.9928ms -> ~71.4651Hz
[TurboJpegRgbPacketProcessor] avg. time: 13.6236ms -> ~73.402Hz
>......

Seems like it is an USB issue.

I tried connecting the kinect to various USB 3.0 ports and 3.1 ports on motherboard and front panel. The outputs turn out to be similar error. 

I also tried running `dmesg`, and this is the output: http://pastebin.com/bgFK2kWX.

Lastly, I ran `ldd | grep libusb` in the depends folder like the Q&A said:
>ldd: missing file arguments
>Try `ldd --help' for more information.

I feel like this is a libusb error, but I don't know how to fix the dependencies and set it up right.

floe, 2015-07-16:
The command in the FAQ was wrong, it should be `ldd ./Protonect | grep libusb` (run this in the bin folder). Also, your PC has an ASMedia USB controller which is known to be incompatible; try to connect it to the Intel controller (maybe use a different USB3 port).

renxinhe, 2015-07-16:
Okay, thanks for the clarification!
Here's what I got:
>$ ldd ./Protonect | grep libusb
>libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f24792d1000)

floe, 2015-07-16:
OK, this is also an issue, because it should point to the custom libusb build in the depends folder. Did you run `depends/install_ubuntu.sh` beforehand?

renxinhe, 2015-07-16:
Yes. I ran `sudo sh install_ubuntu.sh` from /depends, and ldd is printing the same result.

floe, 2015-07-16:
Libfreenect2 _should_ do item # 3 from that list. Apparently, it doesn't seem to work for you, although the output from the script looks correct. You can try to do # 1 yourself: start Protonect as `LD_LIBRARY_PATH=/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/depends/libusb/lib ./Protonect` and see if that changes anything.

renxinhe, 2015-07-16:
Thanks! I ran `sudo LD_LIBRARY_PATH=/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/depends/libusb/lib ./Protonect cpu` in /bin, all 5 windows popped up, and I can see the view from Kinect no problem.

However, when I run it without the `cpu` parameter, the old issue persists:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices

These are all the output. Protonect crashes or something and I'm returned to bash.

Lastly, I tried the `cl` parameter to run with OpenCL. I got this:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 15 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 501314743042
[Freenect2Impl] found 1 devices
X server found. dri2 connection failed! 
Trying to open directly...Device open failed

I have a feeling that my OpenCL still has issues. I have an NVidia GForce 970 GPU. I've already installed the Intel Driver with `apt install beignet-dev`. I'm not sure how to get the NVidia one.

xlz, 2015-07-17:
> Yes. I ran sudo sh install_ubuntu.sh from /depends

Please do not use sudo anywhere during building, installation, and running of Protonect. LD_LIBRARY_PATH is a workaround because somehow it did not work for you. You should be able to link Protonect to local libusb without this hack.

> Protonect crashes or something and I'm returned to bash.

You do have this in your dmesg `Protonect[3410]: segfault at 1c0 ip 00007f11c2bbf388 sp 00007fff52073a70 error 4 in libglfw.so.3.0`. It is OpenGL issue if that's the case.

> I have a feeling that my OpenCL still has issues

OpenCL is not well supported for Nvidia. beignet-dev is Intel only. Your best bet is to use OpenGL with Nvidia driver.

> I've just installed the Nvidia driver with sudo add-apt-repository ppa:ubuntu-x-swat/x-updates

Ubuntu PPA drivers are not well supported with Nvidia/CUDA toolchains. I suggest installing Nvidia driver together with CUDA Toolkit from https://developer.nvidia.com/cuda-downloads.

renxinhe, 2015-07-17:
Thank you for the answer! I installed CUDA toolkit 7 for Linux 64, rebooted, and now both default Prontonect execution and `cl` tag are functional. I didn't install Bumblebee though. I attempted to install it. However, rebooting broke my graphics driver, and I had to reinstall Ubuntu again. Turn out that the CUDA toolkit is enough to update both graphics driver and OpenGL.

>Please do not use sudo anywhere during building, installation, and running of Protonect. 

I had to use `sudo`. Without `sudo`, I get the no device found error:
>[Freenect2Impl] enumerating devices...
[Freenect2Impl] 13 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @4:3!
[Freenect2Impl] found 0 devices
no device connected!

Even after I followed the FAQ, and copied `90-kinect2.rules` into `/etc/udev/rules.d/`, I still have to use `sudo` in order to make the device discovered. 

>You should be able to link Protonect to the local libusb without this hack.

Sorry, but I'm not sure how to do that. :\

xlz, 2015-07-18:
Without bumblebee, it is most likely you are running OpenGL with Intel graphics instead of Nvidia. Installing a wrong configuration would indeed break your graphics. I had it working so you should be able to.

> I had to use sudo. Without sudo, I get the no device found error:

This means usb permissions are not properly set up. Can you provide output of `lsusb` and `ls -l /dev/bus/usb/*/*`?

The temporary way to set up permissions is really `chmod a+w /dev/bus/usb/xxx/yyy` where xxx and yyy come from `lsusb`'s output of `Bus xxx Device yyy`. Running Protonect with sudo is bad. Maybe there needs to be one more documentation fix.

>> You should be able to link Protonect to the local libusb without this hack.

>Sorry, but I'm not sure how to do that. :\

Because I was able to build it with almost identical setup (Ubuntu 14.04, Intel/Nvidia graphics). My only guess is sudo changed installation behaviors. Can you post the output of `cmake ..` of a clean build so we can see how libusb is configured?

renxinhe, 2015-07-18:
Output of `lsusb`:
>Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 22d4:2100  
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 045e:02d8 Microsoft Corp. 
Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 002: ID 17ef:602d Lenovo 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

----------------------------------------------------------------------------------------------------------------------------------------
Output of `ls -l /dev/bus/usb/*/*`:
>crw-rw-r-- 1 root root 189,   0 Jul 18 14:30 /dev/bus/usb/001/001
crw-rw-r-- 1 root root 189,   1 Jul 18 14:30 /dev/bus/usb/001/002
crw-rw-r-- 1 root root 189, 128 Jul 18 14:30 /dev/bus/usb/002/001
crw-rw-r-- 1 root root 189, 129 Jul 18 14:30 /dev/bus/usb/002/002
crw-rw-r-- 1 root root 189, 256 Jul 18 14:30 /dev/bus/usb/003/001
crw-rw-r-- 1 root root 189, 257 Jul 18 14:30 /dev/bus/usb/003/002
crw-rw-rw- 1 root root 189, 258 Jul 18 15:59 /dev/bus/usb/003/003
crw-rw-r-- 1 root root 189, 384 Jul 18 14:30 /dev/bus/usb/004/001
crw-rw-rw- 1 root root 189, 385 Jul 18 15:59 /dev/bus/usb/004/002
crw-rw-rw- 1 root root 189, 386 Jul 18 15:59 /dev/bus/usb/004/003
crw-rw-r-- 1 root root 189, 512 Jul 18 14:30 /dev/bus/usb/005/001
crw-rw-r-- 1 root root 189, 513 Jul 18 14:30 /dev/bus/usb/005/002
crw-rw-r-- 1 root root 189, 640 Jul 18 14:30 /dev/bus/usb/006/001

----------------------------------------------------------------------------------------------------------------------------------------
Output of `cmake CMakeLists.txt`:
>-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Failed
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.0.4
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so  
Linking with these libraries: opencv_videostab;opencv_video;opencv_ts;opencv_superres;opencv_stitching;opencv_photo;opencv_ocl;opencv_objdetect;opencv_ml;opencv_legacy;opencv_imgproc;opencv_highgui;opencv_gpu;opencv_flann;opencv_features2d;opencv_core;opencv_contrib;opencv_calib3d;opencv_videostab;opencv_video;opencv_ts;opencv_superres;opencv_stitching;opencv_photo;opencv_ocl;opencv_objdetect;opencv_ml;opencv_legacy;opencv_imgproc;opencv_highgui;opencv_gpu;opencv_flann;opencv_features2d;opencv_core;opencv_contrib;opencv_calib3d;usb-1.0;/usr/lib/x86_64-linux-gnu/libturbojpeg.so.0;pthread;glfw;/usr/lib/x86_64-linux-gnu/libGL.so;/usr/lib/x86_64-linux-gnu/libOpenCL.so
[... 24 lines cut ...]
[ 65%] Building CXX object CMakeFiles/freenect2.dir/src/resource.cpp.o
[ 69%] Building CXX object CMakeFiles/freenect2.dir/src/command_transaction.cpp.o
[ 73%] Building CXX object CMakeFiles/freenect2.dir/src/registration.cpp.o
[ 78%] Building CXX object CMakeFiles/freenect2.dir/src/libfreenect2.cpp.o
[ 82%] Building CXX object CMakeFiles/freenect2.dir/src/tinythread/tinythread.cpp.o
[ 86%] Building C object CMakeFiles/freenect2.dir/src/flextGL.c.o
[ 91%] Building CXX object CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o
[ 95%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp:48:21: fatal error: CL/cl.hpp: No such file or directory
 #include < CL/cl.hpp >
                     ^
compilation terminated.
make[2]: *** [CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

I've never had issue with finding this header file before during build. Everything here is from a clean github clone. I simply followed the installation guide:
>$ git clone https://github.com/OpenKinect/libfreenect2.git
$ sudo apt-get install build-essential libturbojpeg libjpeg-turbo8-dev libtool autoconf libudev-dev cmake mesa-common-dev freeglut3-dev libxrandr-dev doxygen libxi-dev libopencv-dev automake
$ cd libfreenect2/depends
$ sh install_ubuntu.sh
$ sudo dpkg -i libglfw3*_3.0.4-1_*.deb
$ cd ../examples/protonect/
$ cmake CMakeLists.txt
$ make && sudo make install

xlz, 2015-07-19:
>Bus 004 Device 003: ID 045e:02d8 Microsoft Corp. 
>Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 
>Bus 003 Device 003: ID 045e:02d9 Microsoft Corp. 
>crw-rw-rw- 1 root root 189, 258 Jul 18 15:59 /dev/bus/usb/003/003
>crw-rw-rw- 1 root root 189, 385 Jul 18 15:59 /dev/bus/usb/004/002
>crw-rw-rw- 1 root root 189, 386 Jul 18 15:59 /dev/bus/usb/004/003

Permissions are all right. Protonect should run without sudo. Does it run well?

> Linking with these libraries: 
> ... usb-1.0

This is all right. Now you should have 
```
$ ldd Protonect|grep usb
	libusb-1.0.so.0 => /home/xlz/src/libfreenect2/depends/libusb/lib/libusb-1.0.so.0 (0x00007f77fa550000)
```
Is this right?

> CL/cl.hpp: No such file or directory
> I simply followed the installation guide:

Have you installed `opencl-headers`? It is in this section: "3. OpenCL dependency".

OK. Here is a corner case bug where the user does not want OpenCL but the system supports OpenCL with no `CL/cl.hpp`. The lack of `CL/cl.hpp` does not affect OpenCL detection right now but it causes FTBFS. This could be handled better.

renxinhe, 2015-07-19:
Where should I run `ldd Protonect | grep usb`? 

----------------------------------------------------------------------------------------------------------------------------------------
>Have you installed opencl-headers? It is in this section: "3. OpenCL dependency".

My bad. I forgot that step. I've now installed `opencl-headers`.

---------------------------------------------------------------------------------------------------------------------------------------
I still have compilation errors during `make`, although it appears to be different:
>$ make && sudo make install
[  4%] Built target generate_resources_tool
[  8%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp: In member function ‘void libfreenect2::OpenCLDepthPacketProcessorImpl::listDevice(std::vector<cl::Device>&)’:
/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp:254:12: error: ‘CL_DEVICE_TYPE_CUSTOM’ was not declared in this scope
       case CL_DEVICE_TYPE_CUSTOM:
            ^
/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLDepthPacketProcessorImpl::initDevice(int)’:
/home/jim/UC_Berkeley/Summer_2015/OpenKinect/libfreenect2/examples/protonect/src/opencl_depth_packet_processor.cpp:336:14: error: ‘CL_DEVICE_TYPE_CUSTOM’ was not declared in this scope
         case CL_DEVICE_TYPE_CUSTOM:
              ^
make[2]: *** [CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xlz, 2015-07-19:
> CL_DEVICE_TYPE_CUSTOM’ was not declared in this scope

Nvidia's OpenCl is poorly supported.

Apply this commit https://github.com/xlz/libfreenect2/commit/3401e9b9cff46a2a3ebca08e3d9ff00722aed390

renxinhe, 2015-07-19:
Okay. That commit fixes it!
Thank you so much!

closed 2015-11-09 by floe