OpenKinect/libfreenect2 #322: Segmentation fault in glfGetWindowAttrib

MichielDerhaeg, 2015-07-15:
If I try to run anything I get a segmentation fault, this is what I got running the Protonect example.
It seems that a nullpointer it getting passed around.
````
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff63de8ea in glfwGetWindowAttrib (handle=0x0, attrib=139266) at /home/michiel/libfreenect2/depends/glfw_src/src/window.c:560
560	            return window->glMajor;
(gdb) backtrace
#0  0x00007ffff63de8ea in glfwGetWindowAttrib (handle=0x0, attrib=139266) at /home/michiel/libfreenect2/depends/glfw_src/src/window.c:560
#1  0x00007ffff7a16094 in flextInit (window=0x0, bindings=0x656d00) at /home/michiel/libfreenect2/examples/protonect/src/flextGL.c:20
#2  0x00007ffff7a19d2a in libfreenect2::OpenGLDepthPacketProcessorImpl::initialize (this=this@entry=0x653e10) at /home/michiel/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:472
#3  0x00007ffff7a17b29 in libfreenect2::OpenGLDepthPacketProcessor::OpenGLDepthPacketProcessor (this=0x644480, parent_opengl_context_ptr=<optimized out>, debug=<optimized out>)
    at /home/michiel/libfreenect2/examples/protonect/src/opengl_depth_packet_processor.cpp:786
#4  0x00007ffff7a05366 in libfreenect2::OpenGLPacketPipeline::createDepthPacketProcessor (this=0x634360) at /home/michiel/libfreenect2/examples/protonect/src/packet_pipeline.cpp:109
#5  0x00007ffff7a05646 in libfreenect2::BasePacketPipeline::initialize (this=this@entry=0x634360) at /home/michiel/libfreenect2/examples/protonect/src/packet_pipeline.cpp:43
#6  0x00007ffff7a05a33 in libfreenect2::OpenGLPacketPipeline::OpenGLPacketPipeline (this=0x634360, parent_opengl_context=<optimized out>, debug=<optimized out>)
    at /home/michiel/libfreenect2/examples/protonect/src/packet_pipeline.cpp:102
#7  0x00007ffff7a0fe7f in libfreenect2::createDefaultPacketPipeline () at /home/michiel/libfreenect2/examples/protonect/src/libfreenect2.cpp:627
#8  0x00007ffff7a12841 in libfreenect2::Freenect2::openDevice (this=0x7fffffffd7c0, serial="500195143042") at /home/michiel/libfreenect2/examples/protonect/src/libfreenect2.cpp:760
#9  0x0000000000403744 in main (argc=<optimized out>, argv=<optimized out>) at /home/michiel/libfreenect2/examples/protonect/Protonect.cpp:113
````

floe, 2015-07-15:
`flextInit (window=0x0, bindings=0x656d00)` seems to indicate your OpenGL isn't setup properly. Can you run `glxinfo | grep OpenGL`?

MichielDerhaeg, 2015-07-15:
````
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.1 (Core Profile) Mesa 10.1.3
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 10.1.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
````

floe, 2015-07-15:
The OpenGL DPP needs at least OpenGL 3.3, yours is still at 1.4. Either try upgrading your Xorg stack, or try the OpenCL depth packet processor (`Protonect cl`), although you will probably also need to update some packages for that to work.

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe