OpenKinect/libfreenect2 #325: Get the real distance

yuanmingze, 2015-07-15:
Hi:
1.How can I get or compute real distance from libfreenect2 with Kinect V2 ?

2.On Protonect.cpp
```
cv::imshow("depth_processed", cv::Mat(depth->height, depth->width, CV_32FC1, depth->data) / 4500.0f);
```
Why is divided by 4500.0f?  If I change MaxDepth or MinDepth , whether modify it?

3.Where can I get Kinect V2 specification or Datasheet?

xlz, 2015-07-15:
1. the depth image provides the real distance.
2. 4500.0 millimeter is the maximum distance per spec. Using this value here just makes the image visualize better for any maxDepth by using a sane default normalizing factor. 
3. I don't think there are any datasheets published.

yuanmingze, 2015-07-16:
Thanks a lot.

closed 2015-07-16 by yuanmingze