OpenKinect/libfreenect2 #327: I want to run with openGL but...

Guhnoo, 2015-07-18:
Hello,   I check  It runs well  when I run   './Protonect cpu '   (rgb,depth, etc)

but, I  hope also run  on './Protonect gl ' 

I try to so many times and I guess  My problem is optimus feature. 

#installed bumblebee 
I have installed ' https://wiki.ubuntu.com/Bumblebee#Advanced_Setups ' this. 

and but, I don't know how I have to do next?   

after installed, I get nvidia 304.125 .

larshg, 2015-07-22:
Maybe you can force it to use the gpu in nvidia's settings, or maybe use the beignet project for running it on the cpu if it defaults to this one. But what error are you getting? black screen? What if you run ./protonect cl ? what does it report as devices available?

christiankerl, 2015-07-23:
did you try ```optirun ./Protonect gl```?

Guhnoo, 2015-07-24:
Thanks to answer everyone.

I succeeded to force run like 
## $ optirun ./Protonect
[image]

well, but I have another problem in useing OpenCL 

when I try to calibration in this git hup =>  https://github.com/code-iai/iai_kinect2 
(I will use this in ros)
I have some problem to getting depth img ,trying to use $ optirun --no-failsafe ~~~~~   
[image]

I thought It is because Depth registration method does not change from CPU to OpenCL
So, I followed this  to install beignet
[image]

and rebuild , as trying same section 4 

But, I get error  when I use $ ./Protonect cl
[image]
What is this problem???

floe, 2015-07-24:
This is addressed in the FAQ, when you want to use OpenCL on an Intel GPU, you have to execute `echo 0 | sudo tee /sys/module/i915/parameters/enable_cmd_parser` first if this error occurs.

Guhnoo, 2015-07-25:
Thanks to give me an answer. I fixed it . Really Thanks

closed 2015-07-25 by floe