OpenKinect/libfreenect2 #329: Access to gravity vector / IMU

kevinkarsch, 2015-07-19:
Does anyone have any leads on querying the current gravity vector from the device (or any IMU data for that matter)? In the Kinect SDK, the gravity vector can be found with the following: 

    IBodyFrame* pBodyFrame = NULL;
    HRESULT hr = m_pBodyFrameReader->AcquireLatestFrame(&pBodyFrame);
    Vector4 planeNormal;
    pBodyFrame->get_FloorClipPlane(&planeNormal);
    //Gravity vector is then (-planeNormal.x, -planeNormal.y, -planeNormal.z)

I've tried tracing out the BodyFrame stream, but no luck yet. I have a feeling this data might even be elsewhere, e.g. general IMU readings that aren't part of the BodyFrame.

tlind, 2015-07-19:
I believe the API for the onboard accelerometer is (not) yet exposed, see this discussion in the official forums:
https://social.msdn.microsoft.com/Forums/en-US/97d63a7f-6f73-44ba-a0bb-bd6f47df0a50/i-want-to-read-the-values-of-the-inside-accelerometer-on-kinect-for-windows-v2-sensor?forum=kinectv2sdk

See also issue #175.

floe, 2015-07-20:
Even if it's not exposed by the official SDK, the data is probably still somewhere in the USB data stream. However, this will involve capturing the raw stream and somehow identifying the relevant command being sent...

kevinkarsch, 2015-07-21:
Thanks -- decoding the data stream was exactly what I had in mind. I assume it's one of the ISOCH transfers coming in from the device, but I still haven't been able to figure out which is which. My guess is that the larger ones are associated with color/depth/ir data and the smaller ones are audio. Maybe the "body frame" is its own stream, or this information is possibly encoded somewhere in the other streams. For now, I'll keep looking through and share any updates as I have time, and it would be great if anyone had any insight to share on where the data might be hidden.

floe, 2015-07-21:
AFAIK the body frame is calculated on the host computer based on the raw depth data. You might want to have a look at `include/libfreenect2/protocol/command*.h` and `response.h`, which contain what is publicly known about the protocol.

kevinkarsch, 2015-07-22:
Got it, thanks. I'll report any progress I make here.

xlz, 2016-01-27:
Merged to #175.

closed 2016-01-27 by xlz