OpenKinect/libfreenect2 #330: Bug: VS 2012 (Express) build failures

AliShug, 2015-07-20:
I've had a few issues with getting the build to go through using Visual Studio Express 2012 - I would have changed environments, but didn't have that option:

* Trying to link with OpenCV 3.0 resulted in a notfound error in the first stage of the cmake build. I don't know if this is expected or if I screwed up the installation, either way switching to my existing 2.4 directory seemed to solve this.

* Libusb wasn't correctly linked. The script tried to link:
  `C:/Users/.../depends/libusb/x64/Release/lib` (the directory)
  When it needed:
  `C:/Users/.../depends/libusb/x64/Release/lib/libusb-1.0.lib`

* I had to manually specify the TurboJPEG include and library variables. Not sure what went wrong here, the variables appeared but were set with error values.

* My version of MSVC has terrible C99 support, so doesn't define `roundf`. I had to insert a replacement function into registration.cpp (Registration::Registration uses it a few times)

xlz, 2015-07-21:
> OpenCV 3.0

OpenCV 3.0 has API incompatibilities. We are in the process of removing OpenCV altogether. Please use OpenCV 2.x for the moment.

> Libusb wasn't correctly linked

Need CMake output for this.

> had to manually specify the TurboJPEG
> variables appeared but were set with error values.

What are the wrong values? Again, CMake output please. 

> My version of MSVC has terrible C99 support, so doesn't define roundf

@wiedemeyer I think I have seen one of your patch doing rounding with `(int)(x + 0.5f)`. Shouldn't it already replace all `roundf()`?

AliShug, 2015-07-21:
> Please use OpenCV 2.x for the moment.

No problem, but it might be worth mentioning that in the Readme?

I'm away from my workstation at the moment but I'll try a clean re-build when I get the chance, and post the CMake output here. Anyway, it seemed like it couldn't find the TurboJPEG include and library directories from the provided root, but that might well have been a screw-up on my part; the LibUSB error was much stranger.

There's `roundf()` use in [registration.cpp](https://github.com/OpenKinect/libfreenect2/blob/master/examples/protonect/src/registration.cpp#L244-L258) at lines 244, 245 and 258

AliShug, 2015-07-22:
Okay I've tried again from scratch, and it looks like the build problems were mostly errors on my part; I accidentally built the static libusb library. Don't know what happened with TurboJPEG though.

Apologies for the red herrings - that said, the Readme's build instructions could do with fixing:

```bash
cd example\protonect
mkdir build && cd build
cmake .. -G "Visual Studio 12 2013 Win64" -DCMAKE_INSTALL_PREFIX=. -DOpenCV_DIR=%cd%\..\..\..\depends\opencv\build 
cmake --build . --config Release --target install
```

Should be `cd examples\protonect`, and `-DOpenCV_DIR=..\..\depends\opencv\build`.

xlz, 2015-07-22:
> cd examples\protonect, and -DOpenCV_DIR=..\..\depends\opencv\build

What does this fix? It builds in the build directory to not pollute source directory. Also, `%cd%` is required for specifying paths.

AliShug, 2015-07-22:
> It builds in the build directory to not pollute source directory

Not saying it shouldn't! `example` appears to be a typo, the directory has an 's' on the end. `%cd%` isn't required, at least on my machine, but on trying it again you're right - it does work.

Guess I've just been an idiot!

larshg, 2015-07-22:
You should also be able to use opencv 3.0.0 - atleast I'm using it. Currently using VS2013 or VS2015 community edition, but I used the express version earlier - so that shouldn't be a problem.

It might be that the path should be including ´´´install´´´ also, if you have build opencv yourself? ie. ´´´depends\opencv\build\install´´´ - and its not required to be placed in depends, if you already have it installed somewhere else :)

AliShug, 2015-07-23:
Figured out what was going wrong with the OpenCV path:

```bash
cmake .. -G "Visual Studio 12 2013 Win64" -DCMAKE_INSTALL_PREFIX=. -DOpenCV_DIR=%cd%\..\..\..\depends\opencv\build
```

This doesn't correctly deal with paths containing spaces. On moving to a local directory "Local Documents" I get this error when trying to build:

```bash
CMake Error: The source directory "C:/Users/aw12700/Local Documents/libfreenect2/examples/protonect/build/Documents/libfreenect2/depends/opencv/build" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
```

It's dropped the first part of the `%cd%` path because of the space, then just concatenated what was left to the current directory giving a garbage path.

> You should also be able to use opencv 3.0.0

Alright, I'll give it another go and try to see what was going wrong!

AliShug, 2015-07-23:
Trying to build with OpenCV 3.0 goes through config fine, but breaks on linking.

```
"C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\install.vcxproj" (default target) (1) ->
"C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\ALL_BUILD.vcxproj" (default target) (2) ->
"C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\Protonect.vcxproj" (default target) (3) ->
"C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj" (default target) (5) ->
(Link target) ->
  opencv_videostab300.lib(global_motion.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_videostab300.lib(opencv_videostab_pch.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_videostab300.lib(optical_flow.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_videostab300.lib(outlier_rejection.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_video300.lib(lkpyramid.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_video300.lib(opencv_video_pch.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
  opencv_video300.lib(opencl_kernels_video.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in transfer_pool.obj [C:\Users\aw12700\Local Documents\libfreenect2\examples\protonect\build\freenect2shared.vcxproj]
ETC...........
```

Seems like it's trying to link statically (?) but I don't know what to change to get around that :(
Obviously I can just use 2.x, but I'm curious how you got it working?

larshg, 2015-07-23:
Hmm. I manually build opencv with shared libs, but try the suggestions in #340.

Is it prebuild opencv you use?

AliShug, 2015-07-24:
I've been using pre-built - I'll check with manually built if I get the chance.

The paths problem is admittedly pretty minor, and yeah, it's fixed by adding quotes. Again it might be worth updating the Readme, but if you're removing OpenCV these are all going to be short-lived problems regardless.

xlz, 2015-08-19:
Now OpenCV is removed. Do you still have issues?

floe, 2015-11-09:
Is this fixed with 0.1-rc1? If not, please re-open.

closed 2015-11-09 by floe