OpenKinect/libfreenect2 #331: libusb issue applicable to original libfreenect?

ahundt, 2015-07-20:
I noticed this [new merge dealing with libusb issues on libfreenect2](https://github.com/xlz/libfreenect2/commit/8483bf8622e79e755e5417f3795a34839b0b5ea8). 
Does this libusb issue also possibly apply to the original libfreenect on OS X? I'm having some difficulty over there and if those changes might help I'd be interested in bringing the information over.

floe, 2015-07-20:
No, because the patches mentioned there are specific to USB3, which the Kinect v2 requires. The Kinect v1 is fine with USB2, and therefore these patches won't have any direct effect. OTOH, you _can_ of course try to build a more recent version of libusb and see if that fixes your issues. But the specific commit we select for libusb here won't change anything on its own.

closed 2015-07-20 by floe

ahundt, 2015-07-20:
Ok, thanks! those were the details I was hoping for