OpenKinect/libfreenect2 #333: [Question] Depth feed data format is not as expected?

chipweinberger, 2015-07-21:
I've been doing my best to figure this issue out but am coming up short. I really appreciate any help someone can offer.

I am trying to render the depth feed data using DirectX. My shader code is really simple (isometric projection - then viewed from a oculus rift with perspective), but the output is not right. I'm trying to render it as a bitmap like structure, but there are vast areas of zero values, and the image that is there is repeated three times. huh??
[image]

KINECT_VS_OUTPUT KINECT_VS(KINECT_VS_INPUT Input)
{
	KINECT_VS_OUTPUT Output;

	int x = Input.vertexId % 512;
	int y = Input.vertexId / 512;

	float4 pos = float4((x / 512.0f)-.05f, (y / 512.0f)-.05f, (-1.0f * Input.depth / 1000.0f) - 0.2, 1.0f);
	pos = mul(pos, g_mWorld);
	pos = mul(pos, g_mView);
	Output.vPosition = mul(pos, g_mProj);

	return Output;
}

My buffers are laid out as such....

	D3D11_INPUT_ELEMENT_DESC standard_input_layout[] = {
				{ "Position", 0, DXGI_FORMAT_R32_FLOAT, 0, 0, 				D3D11_INPUT_PER_VERTEX_DATA, 0 },

and updated like this

	// updating the depth buffer
	unsigned char* data = cg.current_depth_frames.at(key)->data;//access the obfreenect2::frame
	update_vector_data((float*) data, renderer);

Thanks for any help =)

xlz, 2015-07-21:
```
float4 pos = float4((x-cx)/fx*z, (y-cy)/fy*z, z, 1.0f);
```

chipweinberger, 2015-07-21:
Thanks for your input and help but...I just tried this and it did not work even more so than mine.....

xlz, 2015-07-21:
OK you also had `g_mView` `g_mProj` matrices doing things here. Make sure they are correct.

Anyway the only obvious error was `y/512.0f` because the maximum of y is 412. I'm not sure why you used 0.05 after that.

chipweinberger, 2015-07-21:
Are you able to confirm that the depth data stored at libfreenect2::frame->data is a array of 32 bit floats, one float per pixel?

xlz, 2015-07-21:
Yes.

chipweinberger, 2015-07-22:
I found the error. Apparently in addition to the Input Layour Desc you must also specify stride when you IASetVertexBuffer(). I had the wrong stride value =)

closed 2015-07-22 by floe