OpenKinect/libfreenect2 #335: Can we use it With Windows 7?

ihsanafredi, 2015-07-21:
Is there any solution to use Kinect v2 with windows 7?
Regards
Ihsan

xlz, 2015-07-22:
Yes. libfreenect2 is the solution to use Kinect v2 with windows 7.

ihsanafredi, 2015-07-22:
thanks for your reply. Really thats great. But which SDK i have to install? As when i installed Kinect SDK v2. It give me error, that minimum windows 8 is required. And does not go further. Which version of SDK should i install?

xlz, 2015-07-22:
Have you tried building libfreenect2 without installing Kinect 2 SDK?

ihsanafredi, 2015-07-22:
No not yet. Should I try witout that? But on github, they say "If you have the Kinect for Windows v2 SDK, install it first. You don't need to uninstall the SDK or the driver before doing this procedure."
So should i try?

HenningJ, 2015-07-22:
That part of the readme is confusing. If you're on windows 7, just ignore that sentence.
Just follow the rest of the build instructions.

ihsanafredi, 2015-07-22:
Its really is confusing. I installed opencv. I am confused in next step, I have Nvidia GPU , so should i install OpenCL? And what to do after that? 
Currently i did cmake but got errors in it, hwo to fix it, or can one guide me what to do after installing opencv. 
Performing C++ SOURCE FILE Test LIBFREENECT2_THREADING_STDLIB failed with the following output:
Change Dir: D:/libfreenect2-master/libfreenect2-master/examples/protonect/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe" "cmTryCompileExec848607458.vcxproj" "/p:Configuration=Debug" "/p:VisualStudioVersion=12.0"
Microsoft (R) Build Engine version 12.0.31101.0

[Microsoft .NET Framework, version 4.0.30319.18444]

Copyright (C) Microsoft Corporation. All rights reserved.

Build started 7/22/2015 2:25:36 PM.

Project "D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj" on node 1 (default targets).

PrepareForBuild:

  Creating directory "cmTryCompileExec848607458.dir\Debug\".

  Creating directory "D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\Debug\".

  Creating directory "cmTryCompileExec848607458.dir\Debug\cmTryCom.7EE5157B.tlog\".

InitializeBuildStatus:

  Creating "cmTryCompileExec848607458.dir\Debug\cmTryCom.7EE5157B.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.

ClCompile:

  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\x86_amd64\CL.exe /c /Zi /W3 /WX- /Od /Ob0 /D WIN32 /D _WINDOWS /D LIBFREENECT2_THREADING_STDLIB /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"cmTryCompileExec848607458.dir\Debug\\" /Fd"cmTryCompileExec848607458.dir\Debug\vc120.pdb" /Gd /TP /errorReport:queue "D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx"

  Microsoft (R) C/C++ Optimizing Compiler Version 18.00.31101 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  

  cl /c /Zi /W3 /WX- /Od /Ob0 /D WIN32 /D _WINDOWS /D LIBFREENECT2_THREADING_STDLIB /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"cmTryCompileExec848607458.dir\Debug\\" /Fd"cmTryCompileExec848607458.dir\Debug\vc120.pdb" /Gd /TP /errorReport:queue "D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx"

  

  src.cxx

D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx(13): error C2065: 'thread_local' : undeclared identifier [D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj]

D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx(13): error C2144: syntax error : 'int' should be preceded by ';' [D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj]

Done Building Project "D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj" (default targets) -- FAILED.

Build FAILED.

"D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj" (default target) (1) ->

(ClCompile target) -> 

  D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx(13): error C2065: 'thread_local' : undeclared identifier [D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj]

  D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\src.cxx(13): error C2144: syntax error : 'int' should be preceded by ';' [D:\libfreenect2-master\libfreenect2-master\examples\protonect\build\CMakeFiles\CMakeTmp\cmTryCompileExec848607458.vcxproj]

    0 Warning(s)

    2 Error(s)

xlz, 2015-10-13:
Yes, I just tested. libfreenect2 works on Windows 7 and does not need Kinect 2 SDK installed.

The README is meant for scenario when Kinect 2 SDK is already installed.

vinjn, 2015-10-19:
By the way, I don't think you can install Kinect 2 SDK on Windows 7. It's only supported after Windows 8.

closed 2016-01-27 by xlz