OpenKinect/libfreenect2 #336: [Issue] [multiple kinects] open(idx) always returns "the device may already be open" even for unique indexes

chipweinberger, 2015-07-22:
Hi =)

First off thank you for this wonderful software. I wish there was a donation button for github ;)

I am trying to get multiple kinects working on windows. First off, I should mention that I am using a November build of libfreenect2 because I was having major build issues and this is one that I found prebuilt.

I have multiple kinects plugged into different controllers. Device enumeration finds both. The first one open(0) opens no issues. open(1) however returns the same kinect handle for some reason....

The offending code seems to be this:

    bool tryGetDevice(libusb_device *usb_device, Freenect2DeviceImpl **device)
    {
        for(DeviceVector::iterator it = devices_.begin(); it != devices_.end(); ++it) {
            if((*it)->isSameUsbDevice(usb_device)) {
                *device = *it;
                return true;
            }
        }

        return false;
    }

I imagine it is important but nevertheless I made it always return false which almost seemed to solve the issue and returned a unique device - but later on the code crashed when trying to call start(). 

Has anyone else run into this issue? Has it maybe been fixed since November? Could anyone maybe provide prebuilt libs for a more recent version? I (embarrassingly) already spend 3 days trying to build the latest version only to run into a litany of different issues ( I got it to build eventually but it was never able to return depth frames. Packets were getting dropped or something like that).

Thanks in advance =)

Best,
Chip

floe, 2015-07-22:
The method to select specific devices has changed quite a bit since November; I would suggest you try and build the current master branch first.

chipweinberger, 2015-07-22:
Okay no worries =) Have you heard of anyone successfully using multiple kinects on windows before? Fingers very crossed I can get it to work.

chipweinberger, 2015-07-22:
Indeed updating to the newest build partially solved my issue but ultimately it's failing where the last one failed as well. 

It appears that it is having trouble communicating with the second device over usb, resulting in access violations when it expects a parameter to be there, but failed. 

For reference here are the log details when starting a second deivce:

	[Freenect2DeviceImpl] starting...
	[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[RgbPacketStreamParser::onDataReceived] skipping rgb packet!
	[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
	
	[CommandTransaction::send] sent number of bytes differs from expected number! ex
	pected: 20 got: 0
	[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
	O
	[CommandTransaction::send] sent number of bytes differs from expected number! ex
	pected: 20 got: 0
	[Freenect2DeviceImpl] ReadData0x14 response
	0 bytes of raw data
	[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
	O
	[DepthPacketStreamParser::onDataReceived] skipping depth packet
	[CommandTransaction::send] sent number of bytes differs from expected number! ex
	pected: 24 got: 0
	[Freenect2DeviceImpl] serial number reported by libusb 502233641942 differs from
	 serial number ²²²²½½½½½½½½½½½½½½½½ε■ε■ε■ε■ε■ε■ in device protocol!
	[CommandTransaction::send] bulk transfer failed! libusb error -1: LIBUSB_ERROR_I
	O
	[CommandTransaction::send] sent number of bytes differs from expected number! ex
	pected: 24 got: 0

chipweinberger, 2015-07-24:
Closing this because updating to the newest build fixed the initial bug. The eventual bug (crashing after starting two devices) can be worked around by using a separate process of libfreenect for each kinect as indicated in another issue.

closed 2015-07-24 by chipweinberger