OpenKinect/libfreenect2 #337: How to transform a RGB image pixel location to a Depth image location?

yuanmingze, 2015-07-22:
libfreenect2's  class of Registration transform  RBG image(1920*1080) to Depth image(512*424).
Geting offset of Depth image from "map_c_off"(registration.cpp)  isn't right .

floe, 2015-07-22:
Currently, there's no function to directly transform from RGB image space to depth image space. However, you can pass an additional `Frame` object of size 1920x1082 (two extra lines!) to the `Registration::apply` method via the `bigdepth` pointer. That frame will then contain the depth data, aligned and upscaled to RGB image space, and you can get the corresponding depth value for a color pixel from there.

yuanmingze, 2015-07-22:
Thanks a lot.

closed 2015-07-22 by yuanmingze

floe, 2015-07-22:
Note: forgot to mention that the 2 extra lines in the `bigdepth` image are basically a 1-pixel border on the bottom and the top, so the actual data is from (0,1) to (1920,1081).

whatisor, 2016-11-11:
Hi, I think you misunderstood Question.
He has a rgb pixel on RAW rgb image (not registered image), he want to get corresponding pixel on depth image or 3d space

rorpis, 2018-12-22:
Currently I'm working on a project were I need to do this in python. How could you do this using the python wrapper? (Get the depth value from the rgb frame, or map a rgb pixel to a depth pixel)