OpenKinect/libfreenect2 #338: Switch off IR light

grigorig, 2015-07-22:
The old Kinect V1 SDK allowed users to (temporarily) switch off the IR light. Does the new Kinect V2 hardware support this as well? The new official SDK unfortunately does not support this. If possible, this functionality should be added to libfreenect2.

floe, 2015-07-22:
If it's not supported by the official SDK, then we don't really have a way to find out if it's possible at all. I assume you mean disabling _just_ the IR light, but keeping the camera running? Based on how the ToF camera works, I'm not quite sure if this would give any picture at all.

floe, 2015-07-22:
According to https://github.com/OpenKinect/libfreenect2/issues/307#issuecomment-116165299, this should be possible after all. You could just try sending this command and see what happens :-)

grigorig, 2015-07-28:
Played around with this to no avail. If I send the 0x0A command after starting streaming,  I don't receive any complete frames anymore. I'll maybe try some other things later.

xlz, 2016-02-04:
Right now I think it's not possible to turn it off outside of the firmware. Every second 30 frames are transmitted from Kinect. Each frame has 10 layers. the 10-th layer is obtained with IR led turned off (by firmware).

floe, 2016-02-08:
IIRC the 10th image isn't used in the depth calculation anyway. Could this simply be exposed as another frame type?

xlz, 2016-02-08:
It could. But what is the use of it?

grigorig, 2016-02-08:
I don't really understand. So the Kinect always sends layer which is always recorded, with the IR LED momentarily turned off? Or what is the deal here?

xlz, 2016-02-08:
The 10th layer can be used to detect special IR blinkers on Xbox One controllers. It might also be used to correct background illumination.

floe, 2016-02-08:
The IR image provided by the depth processor is _without_ ambient illumination, while the 10th image would contain _just_ the ambient illumination. I imagine this might be useful in some computer vision scenarios?