OpenKinect/libfreenect2 #340: [Build][Cmake] Using the build instructions in the wiki Cmake does not build with the same runtime libraries: "error LNK2038: mismatch detected  for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_Dynamic Release' in transfer_pool.obj"

chipweinberger, 2015-07-22:
What is the reason Cmake is trying to use different runtime libraries? Is there a step missing in the build instructions?

Thanks again. Sorry for all the questions. I don't like asking for help typically =P

Also, the other type of build error (which I'm assuming is related) that I am seeing is of this variety:

  libcpmt.lib(xthrow.obj) : error LNK2005: "void __cdecl std::_Xlength_error(ch
ar const *)" (?_Xlength_error@std@@YAXPEBD@Z) already defined in msvcprt.lib(MS
VCP120.dll) [C:\Users\Chip\Desktop\libfreenect rebuild\libfreenect2\examples\pr
otonect\build\freenect2shared.vcxproj]

chipweinberger, 2015-07-22:
For anyone running into this issue - a solution is to only use Cmake to make the VS solution file for you - and use VS to build and linnk. Tpen the libfreenect.sln Cmake made for you. Right click the freenect2 project in the solution explorer->c++->codegeneration->MTd-Debug.

xlz, 2015-07-22:
> What is the reason Cmake is trying to use different runtime libraries?

CMake does not try to use different runtime libraries.

> Is there a step missing in the build instructions?

Likely yes. Right now everything is supposed to be linked with /MD. I don't know how it is possible to link with static OpenCV in your case.

chipweinberger, 2015-07-22:
There is apparently a Cmake flag in the open cv cmake file that lets you specify which runtime libraries to use. You can change it with the cmake-gui as I did. I'm sure there is also a command line option as well.

closed 2015-07-24 by chipweinberger