OpenKinect/libfreenect2 #342: [Runtime Error][Build cuased?] on freshly built protonect. Debug assertion failed. _pFirstBlock == pHead

chipweinberger, 2015-07-22:
Not sure what the source of this error could be but I believe it is a build error.

The build process was to a T and went swimmingly. I built everything as MT_staticDebug in debug mode other than libusb with was built in release mode. I will rebuild libusb in debug an report back.

chipweinberger, 2015-07-22:
Triggered here: protonect 134: std::cout << "device firmware: " << dev->getFirmwareVersion() << std::endl;

chipweinberger, 2015-07-22:
Actually the line before 133: std::cout << "device serial: " << dev->getSerialNumber() << std::endl;

chipweinberger, 2015-07-22:
As indicated here: http://stackoverflow.com/questions/18882760/debug-assertion-failed-expression-pfirstblock-phead

It's possible that compiling with static debug is the problem. I'll try and rebuild with all dynamic debug and report back. The original reason I switched to dynamic was because that is how cmake compiled opencv for some odd reason. Anyone know how to fix that?

chipweinberger, 2015-07-22:
Fixed. The solution was to compile everything with dynamic debug. I Used the OpenCv build tutorial described here http://docs.opencv.org/doc/tutorials/introduction/windows_install/windows_install.html. In the gui there is an option (that pops up after configure is pressed) to compile for static lib and dynamic dll runtime. That is what protonect needs.

I imagine there is a flag in cmake you can also use instead of the gui in order to accomplish this, The build instructions should be updated to reflect this.

AliShug, 2015-07-23:
Yeah I had a similar problem when my build defaulted to static linking. Just stuck with the dynamic (and OpenCV < 3.0)

closed 2015-07-24 by chipweinberger